FreeCAD/FreeCAD #19996: Parameter editing in material editor almost impossible.

RiccaDS, 2025-03-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi 
I was trying to set up a FEM thermomechanical simulation with a few custom materials saved after modifying materials in the default library. Specifically Aluminum-6061-T6 and Steel-X5CrNiMo18-10 but I think this is independent of material. The problem is that I am having a really hard time in modifying the physical parameters  of the materials. I will point to the one I had the most difficulty in setting, **Thermal expansion coefficient**, but I can say I had similar problems also with other parameters.
In the case of CTE, if I double click it's property here
[image]
1-The field becomes automatically zero, which in my opinion is not a good behavior because it prevents from copying or modifying part of the value with unit. 
2-I cannot set values such as:
17.8*10^-6 1/K (this one automatically changes to 0.00)
nor
17.8 µm/m/K (this one automatically changes to 0.00 1/K)
In fact I couldn't understand what format the field would accept. 

I tried in many ways, with different combinations of numbers and units, but the only way I was able to gain a consistent behavior was directly modifying the FCMat files like this:
[image]

### Full version info

```shell
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/140.99
Installed mods: 
  * Curves 0.6.55
  * Manipulator 1.5.7
  * Estimate 0.1.2
  * fasteners 0.5.37
  * ThreadProfile 1.92.0
  * Defeaturing 1.2.3
  * A2plus 0.4.68
```

### Subproject(s) affected?

Material

### Anything else?

I had this discussion in the forum: 
https://forum.freecad.org/viewtopic.php?p=814698#p814698

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RiccaDS, 2025-03-07:
ok. I add one apparent bug also: searchbox does not seem to work for me in material selector. If I write for example "steel" or "aluminum", nothing happens.

FEA-eng, 2025-09-30:
Unit handling might be a bit better now, but most issues still persist. The material editor definitely needs significant improvements and maybe @davesrocketshop will take care of it in the future.

FEA-eng, 2026-10-30:
See [this](https://github.com/FreeCAD/FreeCAD/issues/19996#issuecomment-3353280785)