FreeCAD/FreeCAD #20001: FreeCAD crashes after moving a part in an assembly.

Norbert-Rotorwerk, 2025-03-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FreeCAD crashes after moving a part in an assembly. If you create a connection in the assembly workbench in which you can slide (Assembly_CreateJointSlider) or rotate (Assembly_CreateJointRevolute) one part and you want to move that part FreeCAD crashes immediately if there is a cyclic dependency in the model. 
I have a big assembly model and after insert of some parts the notice "The graph must be a DAG" occur. Whatever, how the cyclic dependency comes, I do not know. The graph diagram is so big and confusing so that I can't find the dependency. In the model there are also assemblies linked to the global assembly. 
My suggestion is it would be good to highlight the cyclic dependency in the dependency diagram. FreeCAD "knows" there is one. And FreeCAD should not crashes. 
How can I add the dependency diagram of my model? The exported version is not allowed to attach. 

Norbert

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * BOLTSFC 2022.11.5
  * parts_library.bak
  * parts_library
  * symbols_library
  * Defeaturing 1.2.0
  * Assembly4 0.12.6
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * Render 2023.2.2
  * Help 1.0.3
  * FEM_FrontISTR 0.2.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.37
  * 3DfindIT 1.2.0
  * sheetmetal 0.4.26
  * BillOfMaterials 1.0.2
  * Design456 0.0.1
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-06:
Could you recreate the issue with a simpler file which you can share as .zip and add steps how to reproduce this? Otherwise it is hard to fix something which is not reproducible.
Please test the file on the dev version if the issue exists there too: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Norbert-Rotorwerk, 2025-03-06:
>>> IMPORTANT <<<
rename the file FreeCAD_Assembly_Crash.zip to FreeCAD_Assembly_Crash.tar than untar the file and you get the file structure of the assembly. I could not make a real zip file because of some names in the file structure. Sorry for that. 

It is a shorter version with the error of a cyclic dependency. 

Steps to reproduce the crash:
1. open the file 'Kipptisch Pitchkonsole Delta4000.FCStd'
2. in the model tree double click on the Assembly to activate the 'Assembly' and load the assembly workbench
3. follow the text in picture 1.png - a result should something like in picture 2.png
4. after trying to move the upper part (marked in picture 1.png) FreeCAD crashes

[image]

[image]

[pictures.zip](https://github.com/user-attachments/files/19108720/pictures.zip)

[FreeCAD_Assembly_Crash.zip](https://github.com/user-attachments/files/19107273/FreeCAD_Assembly_Crash.zip)

luzpaz, 2025-04-06:
@Norbert-Rotorwerk can you kindly add the full [About](https://wiki.freecad.org/About) info of the weekly version you tested on ?

Norbert-Rotorwerk, 2025-04-24:
I have deleted the original version I tested it, but with this version FreeCAD crashes in the same way.

OS: Debian GNU/Linux 12 (bookworm) (MATE/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0197/1
Installed mods: 
  * BOLTSFC 2022.11.5
  * parts_library.bak
  * parts_library
  * symbols_library
  * Defeaturing 1.2.0
  * Assembly4 0.12.6
  * Manipulator 1.5.0
  * MeshRemodel 1.8919.0
  * Help 1.0.3
  * FEM_FrontISTR 0.2.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.38
  * 3DfindIT 1.2.0
  * sheetmetal 0.7.22
  * BillOfMaterials 1.0.2
  * Design456 0.0.1
  * OpenTheme 2024.9.1
  * Render.backup1744616697.684052 2023.2.2 (Disabled)
  * Render

xtemp09, 2025-04-24:
@Norbert-Rotorwerk, _FreeCAD_ fails to open the file you uploaded with the error:

```
 pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Invalid project file /home/xtemp09/Downloads/iss-20001-ass-crash/FreeCAD_Assembly_Crash.FCStd: basic_ios::clear: iostream error
```

It looks like _FreeCAD_ was abruptly terminated when saving, meaning that the file is incomplete.

maxwxyz, 2025-07-15:
@Norbert-Rotorwerk @paranoidi can you reproduce the issue with the new file in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

paranoidi, 2025-07-15:
@maxwxyz crashes with daily build 566fe8371a88938e1b70a6048caac09f345cf024

xtemp09, 2025-07-16:
@maxwxyz, here is the backtrace that paranoidi should have provided. He encountered a duplicate of #20614, as one can see from the video and the backtrace provided below.

<details>

<summary>Backtrace</summary>

```cpp
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff5ca9b82 in __gnu_cxx::__normal_iterator<unsigned long*, std::vector<unsigned long, std::allocator<unsigned long> > >::__normal_iterator (this=0x7fffffffa560, 
    __i=<error reading variable: Cannot access memory at address 0x8>)
    at /usr/include/c++/13/bits/stl_iterator.h:1077
1077          : _M_current(__i) { }
(gdb) bt
#0  0x00007ffff5ca9b82 in __gnu_cxx::__normal_iterator<unsigned long*, std::vector<unsigned long, std::allocator<unsigned long> > >::__normal_iterator (this=0x7fffffffa560, 
    __i=<error reading variable: Cannot access memory at address 0x8>)
    at /usr/include/c++/13/bits/stl_iterator.h:1077
#1  0x00007ffff5ca336c in std::vector<unsigned long, std::allocator<unsigned long> >::end (this=0x0) at /usr/include/c++/13/bits/stl_vector.h:894
#2  0x00007fff9cba93ba in MbD::PartFrame::removeRedundantConstraints (
    this=0x55555d9d3430, 
    redundantEqnNos=std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> >> (empty) = {...})
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/PartFrame.cpp:149
#3  0x00007fff9cba38e1 in MbD::Part::removeRedundantConstraints (this=0x55555d6b6450, 
    redundantEqnNos=std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> >> (empty) = {...})
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/Part.cpp:378
#4  0x00007fff9cbd1f10 in operator() (__closure=0x7fffffffa7e0, 
    item=std::shared_ptr<MbD::Item> (use count 3, weak count 0) = {...})
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/PosICNewtonRaphson.cpp:37
#5  0x00007fff9cbd3764 in std::__invoke_impl<void, MbD::PosICNewtonRaphson::run()::<lambda(std::shared_ptr<MbD::Item>)>&, std::shared_ptr<MbD::Item> >(std::__invoke_other, struct {...} &) (__f=...)
	at /usr/include/c++/13/bits/invoke.h:61
#6  0x00007fff9cbd333d in std::__invoke_r<void, MbD::PosICNewtonRaphson::run()::<lambda(std::shared_ptr<MbD::Item>)>&, std::shared_ptr<MbD::Item> >(struct {...} &) (__fn=...)
    at /usr/include/c++/13/bits/invoke.h:111
#7  0x00007fff9cbd2ff2 in std::_Function_handler<void(std::shared_ptr<MbD::Item>), MbD::PosICNewtonRaphson::run()::<lambda(std::shared_ptr<MbD::Item>)> >::_M_invoke(const std::_Any_data &, std::shared_ptr<MbD::Item> &&) (__functor=..., __args#0=...)
    at /usr/include/c++/13/bits/std_function.h:290
#8  0x00007fff9cc12943 in std::function<void (std::shared_ptr<MbD::Item>)>::operator()(std::shared_ptr<MbD::Item>) const (this=0x7fffffffa7e0, 
    __args#0=std::shared_ptr<MbD::Item> (empty) = {...})
    at /usr/include/c++/13/bits/std_function.h:591
#9  0x00007fff9cc1122b in std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)>) (
    __first=std::shared_ptr<MbD::Part> (use count 3, weak count 0) = {...}, 
    __last=std::shared_ptr<MbD::Part> (use count 1, weak count 0) = {...}, __f=...)
    at /usr/include/c++/13/bits/stl_algo.h:3833
#10 0x00007fff9cc0beef in MbD::System::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) (this=0x55555d9f0730, f=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/System.cpp:161
#11 0x00007fff9cc1a2c1 in MbD::SystemSolver::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) (this=0x55555d981530, f=...)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/SystemSolver.cpp:310
#12 0x00007fff9cbd2186 in MbD::PosICNewtonRaphson::run (this=0x55555914a180)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/PosICNewtonRaphson.cpp:37
#13 0x00007fff9cc19071 in MbD::SystemSolver::runPosIC (this=0x55555d981530)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/SystemSolver.cpp:88
#14 0x00007fff9cc18e66 in MbD::SystemSolver::runAllIC (this=0x55555d981530)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/SystemSolver.cpp:66
#15 0x00007fff9cc0b1df in MbD::System::runKINEMATIC (this=0x55555d9f0730, 
    self=std::shared_ptr<MbD::System> (use count 2, weak count 0) = {...})
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/System.cpp:86
#16 0x00007fff9ca4c472 in MbD::ASMTAssembly::runKINEMATIC (this=0x55555c8aa930)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/3rdParty/OndselSolver/OndselSolver/ASMTAssembly.cpp:1447
#17 0x00007fffa0d3eea3 in Assembly::AssemblyObject::solve (this=0x555559c6e4b0, 
    enableRedo=false, updateJCS=true)
    at /home/fyr/Downloads/FreeCAD/freecad-source/src/Mod/Assembly/App/AssemblyObject.cpp:157
[... cut ...]
Build date: 2025/07/05 20:44:22
Build type: Unknown
Branch: master
Hash: db11f481e9ac74c8f39c513a5cc233058af20fb1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.2463/96.2463/1
```

</details>

PaddleStroke, 2025-07-16:
This was closed because it's a duplicate of 20614?