FreeCAD/FreeCAD #20005: little issue with material editor

Vindex0, 2025-03-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

i have a littel problem which is not always but sometimes when im creating a new material with as example "1.4301 - stainless" at the beginning of the name it removes teh name and just saves "1" ... i found a workaround now when i saved the file i open it with texteditor and change the name by hand ...

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * InventorLoader 1.5.1
  * sheetmetal 0.7.11
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-05:
I can confirm the issue. The file name is created correctly, but the `Name` property of the material is not correct.

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

maxwxyz, 2025-03-05:
@davesrocketshop FYI

Tiago-Almeida007, 2025-03-06:
Hi, is there anyone already working on this bug? If not and would be interested in solving it. Could this issue be assigned to me? I'm very interested in contributing to this repository!

davesrocketshop, 2025-03-06:
I have no objections to you working on it. There is a lot of work that will need to be done on the editor that is waiting on some other merges, but this shouldn't interfere with that work.

Please make sure I have a chance to review before merging to ensure nothing important gets changed accidentally.

commit 9153926c (2025-03-31): fix #20005: material editor issue
-Changed info.baseName() to
info.fileName().remove(QStringLiteral(".FCMat"), Qt::CaseInsensitive)
in MaterialLibrary.cpp to ensure that only the extension ".FCmat"
is removed from the file name bacause the previous version was removing
 everything in front of the first dot
PR #20489 (2025-03-31): Mod: Fix Material Library naming issue
fixes #20005 :
Updated info.baseName() to info.fileName().remove(QStringLiteral(".FCMat"), Qt::CaseInsensitive) in MaterialLibrary.cpp to ensure that only the ".FCMat" extension is removed from the file name.
Previously, the implementation removed everything after the first dot, which could unintentionally modify file names when saving a material. This fix ensures that only the specific ".FCMat" extension is stripped while preserving the rest of the file name.
@davesrocketshop
closed 2025-03-31 by chennes