FreeCAD/FreeCAD #20010: Build 40504 corrupting files when saved, can't load result file

nv-steve, 2025-03-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I noticed that when I open a file last saved with #40317, save it with #40504 the file can no longer be successfully loaded. I've attached an example file that was last saved with 40317.

**Reproduction steps:**

1. Open attached file "bookcase A - carcasses rev35 - panel drill jig.FCStd.tar.gz" in build 40504
2. Note there are no errors when loading the file
3. Save the file with a new name (not critical, but keeps original file in tact)
4. Close the document in FC
5. Open the new file you saved in step #3
6. Observe the errors

**Errors when loading file:**
```
11:37:38  Invalid Document.xml: End of document reached
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAJAiJng/usr/Mod/Assembly/JointObject.py", line 986, in attach
    groundedObj = app_obj.ObjectToGround
                  ^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'ObjectToGround'
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  Template not set for Page
11:37:39  DVP::execute - View020 - Source shape is Null.
11:37:39  DVP::execute - View021 - Source shape is Null.
11:37:39  DVP::execute - View022 - Source shape is Null.
11:37:39  DVP::execute - View020 - Source shape is Null.
11:37:39  DVP::execute - View021 - Source shape is Null.
11:37:39  DVP::execute - View022 - Source shape is Null.
11:37:39  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAJAiJng/usr/Mod/Assembly/JointObject.py", line 877, in getIcon
    if self.app_obj.JointType == "Fixed":
       ^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'JointType'
11:37:39  pyException: Traceback (most recent call last):
  File "/tmp/.mount_FreeCAJAiJng/usr/Mod/Assembly/JointObject.py", line 877, in getIcon
    if self.app_obj.JointType == "Fixed":
       ^^^^^^^^^^^^^^^^^^^^^^
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'JointType'
```

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Anything else?

[bookcase A - carcasses rev35 - panel drill jig.FCStd.tar.gz](https://github.com/user-attachments/files/19095566/bookcase.A.-.carcasses.rev35.-.panel.drill.jig.FCStd.tar.gz)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-03-06:
I cannot reproduce. I'm attaching the file I saved following the steps to reproduce (rename the file by removing the `.zip` extension): 

[bookcase A - carcasses rev36 - panel drill jig.FCStd.zip](https://github.com/user-attachments/files/19100463/bookcase.A.-.carcasses.rev36.-.panel.drill.jig.FCStd.zip)

When comparing the `Document.xml` file inside of each of the `.FCStd` files, this is the only remarkable change I saw (left is original, right is mine):
[image]

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40522 (Git) Snap 1322
Build date: 2025/03/05 00:12:30
Build type: Release
Branch: main
Hash: edca54767b311611362d615ec15dd1907699fd87
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/159.34
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

nv-steve, 2025-03-06:
Interesting. Looks like you are on version 40522 and I'm experiencing the issue with 40504, but I don't know how much that matters, if at all.
I just went through the steps again to make sure I wasn't imagining this. I'm not and I made a video showing what I'm doing:
https://youtu.be/AstIP7lxNHo

I compared my Document.xml before and after and I have 99 differences in mine. I've attached a diff file if that is helpful at all.
[image]

[Document_diff.xml.tar.gz](https://github.com/user-attachments/files/19101314/Document_diff.xml.tar.gz)

furgo16, 2025-03-06:
> Please test in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

I think the OP is already doing that (build 40504 in about info)

maxwxyz, 2025-03-06:
But it seems not present in a newer dev build 40522

nv-steve, 2025-03-07:
> Oh, I wasn't suggesting that you did anything wrong. I simply intended to state that I could not reproduce it on my system. 
Yeah I know, but I still wanted to confirm it was really happening. Sometimes you can be sure of a problem then later you can't reproduce it. It was good to confirm.

> But it seems not present in a newer dev build 40522
When build 40522 is available as an AppImage I can try again and update this Issue. Or do you have another plan?

luzpaz, 2025-03-22:
@nv-steve please retest

WandererFan, 2025-08-19:
This appears to be a duplicate of "Techdraw: PR19753 regression - CosmeticVertex Save/Restore of tags causes load failure #21069".

closed 2025-08-19 by luzpaz