FreeCAD/FreeCAD #20014: Selection highlighting breaks when selecting disallowed object

Jarno-de-Wit, 2025-03-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## Problem summary
The selection highlighting breaks in some cases, when a disallowed object is selected. When this happens, the selection appears empty, as nothing is highlighted anymore, even if it isn't actually empty.
 There are two ways in which this behaviour can be triggered (see below). The bug only seems to trigger if the selecting action is performed from the Gui by the user clicking on the body / feature, but not if the selection is added from the Python console, despite both actions triggering the same callback from the selectionObserver and selectionGate's point of view.

## Reproduction Method 1: Through a SelectionGate
1. Create a SelectionGate which only allows e.g. faces to be selected.
2. Select a face.
3. Multi-Select an edge (clicking on the edge despite the disallowed symbol) (multi-selection can either be done with ctrl, or by changing selection mode to greedy beforehand)
4. The selection appears empty, with no parts being highlighted. Querrying `Gui.Selection.getSelectionEx()` or `Gui.Selection.getCompleteSelection()` however shows that the original face is still selected, but just not highlighted.

Example gate macro:
```py
class Gate():
    def allow(self, *feature):
        return feature[2].startswith("Face")

Gui.Selection.addSelectionGate(Gate())
```

## Reproduction Method 2: Through a selectionObserver callback
1. Create a selectionObserver like the one in the example macro below
2. Select a face.
3. Multi-Select an edge (either by ctrl-clicking, or by changing the selection mode to greedy beforehand)
4. The selection appears empty, but isn't, similar Method 1.

Example observer macro:
```py
class Observer():
    def addSelection(self, *feature):
        if not feature[2].startswith("Face"):
            # Immediately try to remove the selected feature
            Gui.Selection.removeSelection(*feature[:3])

Gui.Selection.addObserver(Observer())
```

### Full version info

```shell
OS: macOS Mojave (10.14)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Render 2024.12.15
  * UVUnwrap
  * A2plus 0.4.68
  * Curves 0.6.58

Also occurs in the most recent weekly build:

OS: macOS Mojave (10.14)
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 72/128
Installed mods: 
  * Render 2024.12.15
  * UVUnwrap
  * A2plus 0.4.68
  * Curves 0.6.58
```

### Subproject(s) affected?

Core

### Anything else?

The video below shows a reproduction of the bug using both of the methods explained above.
https://github.com/user-attachments/assets/65cba117-e7f3-471f-b2ec-2603ef2b75e1

The macros used in the video above:
[Macros.zip](https://github.com/user-attachments/files/19099189/Macros.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-08:
Confirmed. This can also be tested with [Part_SelectFilter](https://wiki.freecad.org/Part_SelectFilter).

luzpaz, 2025-05-11:
Can anyone propose a fix for this ?