FreeCAD/FreeCAD #20029: AddonManager can't update a single WB

Zolko-123, 2025-03-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I need to update en external module because it causes errors and warnings during startup, and the AddonManager sees that this WB has an update (which actually is wrongly shown in the dialog, this is not the latest update), but it wants to update all addons : 

[image]

I only want to update that single WB, but there is no button for that. I don't want to update all addons as experience has shown that updates can break things and I can't afford this to happen now

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40253 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: e16c462916a050909ad107e4f1ac2e0e5990389d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US) [ OS: English/United Kingdom (en_GB) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Logical/physical DPI: 96.3411/157.74
Installed mods: 
  * A2plus 0.4.56a
  * CurvedShapes 1.0.3
  * fcgear 1.0.0
  * sheetmetal 0.2.50
  * TabBar
  * Render 2022.2.0
  * Curves 0.5.1
  * fasteners 0.5.21
  * Assembly4_dev 0.50.18-dev
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

chennes, 2025-03-06:
The remote cache is only updated once per day: is this a recent update you are looking for? I can force the process to run to generate a new cache if so.

Zolko-123, 2025-03-06:
I only want to update a single WB out of 3 possible updates. I'd like there to be a button on the page of each addon that allows to update that particular addon, without having to update all possible updates. Meaning a fourth button "Update" besides the "Check for updates", "Uninstall" and "Disable".

chennes, 2025-03-06:
That button does check for updates: but it's checking a centralized cache that is only generated once per day. Is the update you are expecting to see something in the last 24 hours?

leoheck, 2025-03-07:
@chennes I am afraid that what you are asking doesn't make any sense. Did you understand what is being requested here?

The user should be able to select the update he wants.

The current behavior is to update all or nothing.

For instance, I have 10 things to be updated. But at that time I want to update just a single thing, otherwise I will have 10 things crashing or changing behavior and this is kinda hell if you are actually trying to design something.

chennes, 2025-03-07:
My point is, there is no update to select: the three available updates shown at the bottom do not apply to this addon: there are no available updates for A2plus. that's *why* the button says "Check for updates" instead of "Update" -- if there were updates available, the button would indeed be an Update button, and would do exactly as you ask. My question is directed as ascertaining *why* there are no available updates, since I presume that the whole reason for the query is that @Zolko-123 made an update to the addon, and doesn't see it reflected in the Addon Manager. One possible reason for that is that the update was made very recently, and the server-side cache at https://www.freecad.org/addons/addon_cache.json hasn't been updated since. It's updated by a 24-hour systemd timer.

Zolko-123, 2025-03-07:
@chennes : you are overthinking this. It's much simpler :  I use FreeCAD v0.21.3 (the one that I maintain because FreeCAD refuses to maintain the 0.21 branch) and my installed A2+ version doesn't give any errors. Sometimes I have to fire up later versions to check for what some Asm4 users ask and there A2+ gives me errors : 

```
$ ./FreeCAD_weekly-builds-40504.AppImage
FreeCAD 1.1.0, Libs: 1.1.0devR40504 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Sheet Metal workbench loaded
During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in /home/hubertz/.local/share/FreeCAD/Mod/A2plus/./InitGui.py
Please look into the log file for further information
The `.FCStd#` backup format is deprecated and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError
Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError
Request failed: PySide2.QtNetwork.QNetworkReply.NetworkError.ContentNotFoundError
```

[image]

so I go to the AddonManager and see that there is an update available that might have fixed these errors : I have 0.4.56a installed and latest version is 0.4.68. The v0.21.3 UI nicely highlights this in red. The 1.1-dev version on the other hand tells me that my 0.4.56a version is up-to-date with the available 0.4.68

[image]

And when I click on the A2+ section I have no update button : 

[image]

I have of course updated the local cache, checked for updates many times ... but never an `Update` button

Syres916, 2025-03-07:
We had an issue about 3 months ago and the entire cache had to be deleted. I have a back burner job to create a button in the AddonManager Preferences to _Clear Cache_ but there's a significant number of _way_ higher priority jobs before that I'm afraid.

chennes, 2025-03-07:
@Zolko-123 thanks for the clarification. Are you using the same Mod directory for 1.1 and 0.21, or are you using a separate one for each version? Also, do both versions of FreeCAD show the same addons as having updates, if you filter by "Update available"?

Zolko-123, 2025-03-08:
@chennes : yes I use the same bok standard directories for both versions (Mod, config ...). And the show the same updates when filtering on updates. What's interesting is when filtering on installed WBs : here, 0.21 correctly identifies available updates (even though it doesn't propose to update them) while 1.1 tells that everything is fine even though it lists outdated WBs

[image]

[image]

Zolko-123, 2025-03-08:
Actually : if the AddonManager can't recognize available updates and is incapable of updating installed addons, then this is a bug not a feature request

Zolko-123, 2025-03-08:
> We had an issue about 3 months ago and the entire cache had to be deleted. I have a back burner job to create a button in the AddonManager Preferences to _Clear Cache_ 

there already is a button to update the local cache

> but there's a significant number of _way_ higher priority jobs before that I'm afraid.

like what ? Re-implementing existing features in new and incompatible ways ? Did you ever wonder why FreeCAD is released under the LGPL and not the GPL ? It's because it never was intended to be a standalone software, it was intended to be a collection of addons. Therefore, the AddonManager is the single most important function of FreeCAD !