FreeCAD/FreeCAD #20035: Create Hole Axis crashing Freecad

leoheck, 2025-03-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Create Hole Axis crashing Freecad

If I hover or click on the newly created Hole_Axis, freecad crashes

[cyclic_dep_1.FCStd.zip](https://github.com/user-attachments/files/19114143/cyclic_dep_1.FCStd.zip)

[image]

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45330) [0x77db7b045330]
#1  0x77db56151fc6 in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /tmp/.mount_FreeCAonIhND/usr/lib/./libTKBRep.so.7.7+0x26
#2  0x77db56163b21 in BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) from /tmp/.mount_FreeCAonIhND/usr/lib/./libTKBRep.so.7.7+0xd1
#3  0x77db56164d35 in BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) from /tmp/.mount_FreeCAonIhND/usr/lib/./libTKBRep.so.7.7+0xd5
#4  0x77db603cf67e in Measure::Measurement::length() const from /tmp/.mount_FreeCAonIhND/usr/lib/Measure.so+0x15e
#5  0x77db48cacc0e in MeasureGui::QuickMeasure::printResult() from /tmp/.mount_FreeCAonIhND/usr/lib/MeasureGui.so+0x34e
#6  0x77db48cad8ed in MeasureGui::QuickMeasure::processSelection() from /tmp/.mount_FreeCAonIhND/usr/lib/MeasureGui.so+0x5d
#7  /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x77db7b6eb6f2]
#8  0x77db7b6ef34a in QTimer::timeout(QTimer::QPrivateSignal) from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5+0x2a
#9  0x77db7b6e064d in QObject::event(QEvent*) from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5+0x7d
#10  0x77db7c5703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#11  0x77db7e37f629 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libFreeCADGui.so+0xc9
#12  0x77db7b6b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x77db7b709f51 in QTimerInfoList::activateTimers() from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5+0x381
#14  /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5(+0x30a724) [0x77db7b70a724]
#15  /tmp/.mount_FreeCAonIhND/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x77db79910ad7]
#16  /tmp/.mount_FreeCAonIhND/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x77db79913ac7]
#17  /tmp/.mount_FreeCAonIhND/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x77db799140ce]
#18  0x77db7b70aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5+0x66
#19  0x77db7b6b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5+0x11b
#20  0x77db7b6bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libQt5Core.so.5+0x82
#21  0x77db7e2decad in Gui::Application::runApplication() from /tmp/.mount_FreeCAonIhND/usr/bin/../lib/libFreeCADGui.so+0x70d
#22  /tmp/.mount_FreeCAonIhND/usr/bin/freecad(+0x7906) [0x56f5018f1906]
#23  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x77db7b02a1ca]
#24  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x77db7b02a28b]
#25  /tmp/.mount_FreeCAonIhND/usr/bin/freecad(+0x7c8e) [0x56f5018f1c8e]
```

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.20
  * DynamicData 2.73.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-07:
No datums should not be included in quickmeasure @PaddleStroke FYI

Zolko-123, 2025-03-07:
> Confirmed and educated guess `HoleAxis_1` is a `PartDesign::Line` and the code in QuickMeasure is:
> 
> ```
>     // only measure shapes
>     if (obj && obj->isDerivedFrom<Part::Feature>()) {
> ```
> 
> so should it be measured???

good guess. But wrong question : should `QuickMeasure` be enabled by default ? Or installed ? I never use it, I didn't even know it existed

closed 2025-03-17 by chennes