FreeCAD/FreeCAD #20042: Assembly: Missing Body in "Insert Component" when Body is referencing Assembly

Tabimeter, 2025-03-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi,
I think I've encountered a problem where I'm unable to insert a body into my assembly.
My body (Cover) is made with an **additive pipe** where a sketch (Cover_Sketch) is following a sketched path (Cover_Path). 
I put **datum points** (Cover_Sweep_Point[x]) on some bodies so that I can make a **SubShapeBinder** from the assembly inside my body (Cover) to reference it in my sketch (Cover_Sketch).
When I now try to insert it in the assembly not only my body (Cover) missing, but the whole group (Plates) is missing, including the already inserted body (Instrument_Plate), for insertion.
When I now delete the binder (Cover_Binder) I'm able to insert it in the assembly.

It happens in the Flathub version (also safe mode) as well as in the dev. Version (40504). 
Attached is the .FCStd and a macro

[assembly_bug.zip](https://github.com/user-attachments/files/19124689/assembly_bug.zip)

### Full version info

```shell
OS: KDE Flatpak runtime (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * fasteners 0.5.37
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Tabimeter, 2025-03-07:
After a quick browse through the code I think my problem comes from src/Mod/Assembly/CommandInsertLink.py.

    def buildPartList(self):
        self.form.partList.clear()

        docList = App.listDocuments().values()

        for doc in docList:
            # Create a new tree item for the document
            docItem = QtGui.QTreeWidgetItem()
            docItem.setText(0, doc.Label + ".FCStd")
            docItem.setIcon(0, QIcon.fromTheme("add", QIcon(":/icons/Document.svg")))

            if not any(
                (child.isDerivedFrom("Part::Feature") or child.isDerivedFrom("App::Part"))
                for child in doc.Objects
            ):
                continue  # Skip this doc if no relevant objects

            self.form.partList.addTopLevelItem(docItem)

            def process_objects(objs, item):
                onlyParts = self.form.CheckBox_ShowOnlyParts.isChecked()
                for obj in objs:
                    if obj == self.assembly:
                        continue  # Skip current assembly

                    if obj in self.assembly.InListRecursive:
                        continue  # Prevent dependency loop.
                        # For instance if asm1/asm2 with asm2 active, we don't want to have asm1 in the list

                    if not obj.ViewObject.ShowInTree and not self.showHidden:
                        continue

                    if (
                        obj.isDerivedFrom("Part::Feature")
                        or obj.isDerivedFrom("App::Part")
                        or obj.isDerivedFrom("App::DocumentObjectGroup")
                    ):
                        # Special handling for DocumentObjectGroup: only add if it contains relevant child objects
                        if obj.isDerivedFrom("App::DocumentObjectGroup"):
                            if not any(
                                (
                                    (not onlyParts and child.isDerivedFrom("Part::Feature"))
                                    or child.isDerivedFrom("App::Part")
                                )
                                for child in obj.ViewObject.claimChildrenRecursive()
                            ):
                                continue  # Skip this object if no relevant children

                        if obj.isDerivedFrom("Part::Feature"):
                            if onlyParts:
                                continue  # Ignore solids if we show only Parts

                        # Now add the object under the document item
                        objItem = QtGui.QTreeWidgetItem(item)
                        objItem.setText(0, obj.Label)
                        objItem.setIcon(
                            0, obj.ViewObject.Icon if hasattr(obj, "ViewObject") else QtGui.QIcon()
                        )  # Use object's icon if available

                        if not obj.isDerivedFrom("App::DocumentObjectGroup"):
                            objItem.setData(0, QtCore.Qt.UserRole, obj)

                        if obj.isDerivedFrom("App::Part") or obj.isDerivedFrom(
                            "App::DocumentObjectGroup"
                        ):
                            process_objects(obj.ViewObject.claimChildren(), objItem)

            guiDoc = Gui.getDocument(doc.Name)
            process_objects(guiDoc.TreeRootObjects, docItem)
            self.form.partList.expandAll()

        self.adjustTreeWidgetSize()

maxwxyz, 2025-03-08:
@Tabimeter do you want to create a PR to fix this issue?
@PaddleStroke FYI

Tabimeter, 2025-03-09:
I've looked into it and I think this part is the problem:
```
if obj in self.assembly.InListRecursive:
                    continue  # Prevent dependency loop.
                    # For instance if asm1/asm2 with asm2 active, we don't want to have asm1 in the list
```
It's prohibiting the insertion of objects which are linking to the active assembly.
Also, it's traversing the Doctree from the root (CommandInsertLink.py; line 262) and matches the first occurrence which is where the group problem comes from.
I suggest:
a) on match: traverse the tree deeper to prohibit more precise.
b) allow recursive dependency with a checkbox in the insert menu under "Show only Parts" and "Rigid sub-assemblies" so the user can decide for himself if he wants to point a gun at his feet :D.

Correct me if I'm wrong otherwise i would try to get an PR out next week.

PaddleStroke, 2025-03-10:
Yes you are correct. My suggestion for the circular dependency is that you make a) in a way that requires no 'Recursive dependency' checkbox. The user should not have to deal with such a setting. So you need to come up with a smart way of doing a) that does not involve any shooting :)

FEA-eng, 2026-10-19:
[This](https://github.com/FreeCAD/FreeCAD/issues/20042#issuecomment-4221328541) is still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```