FreeCAD/FreeCAD #20045: Shortcut to Measure things

leoheck, 2025-03-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Can we have a shortcut to measure things? It could be `d` as it is in the generic constraints in Sketcher.

### Full version info

```shell
1.1-dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-07:
You do not need a shortcut for most measurements, just look in the status bar:

[image]

Works also with multiple selected elements.

closed 2025-03-07 by maxwxyz

leoheck, 2025-03-07:
@tarman3  I don't want to create macros.
I am requesting a standard shortcut for the standard way to measure things.
Measuring things is important, a shortcut is necessary.

leoheck, 2025-03-07:
> You do not need a shortcut for most measurements, just look in the status bar:

I don't understand what you are saying. 
"You don't need" what does that mean?