FreeCAD/FreeCAD #20046: The reference to a BaseFeature is removed when the referenced object is gone.

leoheck, 2025-03-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The reference to a BaseFeature is removed when the referenced object is gone.
This is wrong behavior I suppose and helps nothing the user.
It the referenced objects is removed, the reference to it (the value/text) should be kept so the user can locate and fix the issue faster.

[image]

[image]

### Full version info

```shell
1.1-dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-08:
This seems a very strange idea. Do you really want the UI to show a deleted object as if it has not been deleted? How much confusion would that cause? Note that this a core issue that affects all properties that have the same or similar types.

Roy-043, 2025-03-08:
Also note that you must have ignored a warning when you deleted the referenced object.

maxwxyz, 2025-03-09:
@FreeCAD/design-working-group is there an opinion from you?

leoheck, 2025-03-09:
> This seems a very strange idea. Do you really want the UI to show a deleted object as if it has not been deleted? How much confusion would that cause? Please give this some more thought. Note that this is a core issue that affects all properties that have the same type or similar types.

The object was deleted. But it was linked in another place. Lets say, it can be a single place, or more than one.  

Now a single link is broken (or more than one). It doesn't matter, for the observed result, if the reference is kept or cleared. The link will have a broken symbol.

If the reference is kept, the user can locate and fix those link quicker with the knowledge of what was there in the first place.

On other words. The link doesn't change. What happens is that the referenced object doesn't exist anymore.

The link will be broken with or without the reference.

FEA-eng, 2025-09-30:
@leoheck Can you share the file or just steps to reproduce?

leoheck, 2025-09-30:
Sure @FEA-eng:

1. Create the file 1.FCStd
2. Create a Body inside this file.
3. Create the file 2.FCStd
2. Create a Body inside this file.
5. Create another file called Main.FCStd 
6. Create a link to one of those Bodies.
7. (wait some time, hours, days... forget about the linked Body)
8. Make a folder
9. Move the 1.FCStd and 2.FCStd inside this new folder to organize your project.
10. Launch the Main.FCStd
11. Links are broken, but not only that, the fields are empty. The user does not know what was there to fix the issues.

Broken links should not be cleared; otherwise, we lose the reference to what was there in the first place.

FEA-eng, 2025-09-30:
> What about the Linked Object property? Can you click on the Link object to show if it is empty?

Sure, it is indeed empty:

[image]