FreeCAD/FreeCAD #20048: Duplicated file, always after a crash.

leoheck, 2025-03-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Freecad 1.1-dev crashes constantly.

I always launch Freecad from the command line like this:
```
freecad Body.FCStd
```

Then, after a crash, launching it the way above, I always see this issue.
A duplicated file is opened and then the second one, that needs to be saved, is focused.

The duplicate is the file being recovered by Freecad.
The first one is the one I passed from the command line.

[image]

[image]

Can we prevent such a thing?
I think the crashed files should open up first, then if something was passed to the command line, and if they are not yet opened, it should not open it again.

With a single file, this is easy to spot but a pain since Freecad dev crashes multiple times a day.
If I am passing multiple files at once, then it becomes a real pain.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.20
  * DynamicData 2.73.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-08:
Can it be that this is related to the use of Dropbox files?

leoheck, 2025-03-09:
@Roy-043 I do use Dropbox, indeed.

FEA-eng, 2025-09-30:
@leoheck Is it still relevant? I checked on Windows and I don't have this issue but it might be related to Linux and, even more likely, the use of Dropbox.