FreeCAD/FreeCAD #20056: BIM: Structure objects do not get nodes

Roy-043, 2025-03-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the current dev version structure elements no longer have their Nodes property specified. This is a regression.

Test:
1. Create a beam by clicking two points on the XY plane.
2. Check the Nodes property of the object (on the Data tab).

### Full version info

```shell
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/86.4394
Installed mods:
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-08:
I can confirm the issue.

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

yorikvanhavre, 2025-07-07:
Sure! Done ;)

semhustej, 2025-10-01:
It seems to me it is fixed.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.3.post2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
  * UpdatedPythonPackages

```

closed 2025-10-02 by Roy-043