FreeCAD/FreeCAD #20057: Subtractive Pipe cannot remove splitter

leoheck, 2025-03-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is happenig a lot. 
Changing the Refine from True to False works every time. 
[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/145.143
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.20
  * DynamicData 2.73.0
  * Curves 0.6.58
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * GDT
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.2
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
Does setting refine to false fix the issue?

leoheck, 2025-03-09:
Yes, in those cases it does fix the issue. However, it has to be changed to False after this problematic feature forever.

- Maybe the Refine property should be kept equal to the last operation for the next Features automatically every time it is changed. 

The workflow to change the Refine options is tricky in this situation since this operation cannot be finished with the Refine True. So, with the Task open, we will click Apply/Ok or whatever to have the Feature applied. But then it will fail, so the Task won't close since it has this particular error. Then the user has to look for the properties (on the other side), to change the Refine to False while the Task is open.

A better workflow could be implemented in these corner cases.

1. The error message itself could indicate the `Refine operation` could pass the issue. 
2. The Refine property itself could be given as a checkbox available in the Task panel itself.

FEA-eng, 2026-03-01:
@leoheck Please share the file if it's still relevant.

closed 2026-03-01 by maxwxyz