FreeCAD/FreeCAD #20062: Moving to version 1.0 has broken plugins

Jimmeee, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have things that were working fine in the version right before 1.0, and now they give me messages like: 

"16:15:34  3DfindIT: FreeCAD below version 0.19 is not supported. Please update to a recent version."

I don't know if this is a real error or the authors of some plugins have just configured their plugins to look only after the decimal place, so it thinks that "1.00" is ".00"

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * 3DfindIT 1.2.0
  * Help 1.0.3
  * FEM_FrontISTR 0.1.0
  * 3D_Printing_Tools
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

FindIt at least, but I have seen plenty of other errors.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-08:
It cannot be the core developers responsibility to get Addon authors to merge Pull Requests, I submitted a PR in September 2024 and the author hasn't communicated so I assume they've abandoned it as there's been no changes in 3 years. I personally compile FreeCAD with WebEngine but it's not part of the project supplied builds as there's been security concerns with it.

Jimmeee, 2025-03-08:
OK, I guess I will just remove plugins that give errors. Is that the best course of action?

closed 2025-03-08 by maxwxyz