FreeCAD/FreeCAD #20063: Fillet an already filleted feature

Jimmeee, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have two features that I want to fillet (radius). If I put one down and fillet it, it works fine. Then I add the second feature, and do a Boolean remove to get the shape I want. This removal operation intersects with the previously-filleted features.

I don't want the edges on the newly-subtracted part to be sharp, so I try to fillet them. They will not allow selection. No matter what I do I get a message saying a face line or vertex has not been selected from an active body (there is only one body there, so this seems impossible). 

Note that I got into writing up this error message because of trying to fillet something that has already been filleted. I thought perhaps it was confusing the fillet routine, so I would try to simplify it and I would create all the features ahead of time (with sharp edges) and then fillet them all at once. This did not work either. Then I just started not being able to select anything at all. 

I am very new to FreeCAD, so it is quite possible that I am doing something wrong, but fillet, or more specifically, the selection of what to be filleted, does not seem to work correctly.

Attached is a cylinder with 7 smaller cylinders through it (to be subtracted) and 2 rectangles (also to be subtracted). I have left the part as it is before the boolean subtraction and also left a copy from after the boolean selection. Everything is correct up to these points. But, if I then want to put radiuses (fillets) on all the holes created by the removal of the cylinders and rectangles, I cannot figure out how to select anything. In fact, when I try to select anything in the pre-boolean shape now and subtract it to try again, the program crashes. 

I have seen similar errors (there are MANY fillet error reports), but none seem to be exactly this:

https://github.com/FreeCAD/FreeCAD/issues/19594
https://github.com/FreeCAD/FreeCAD/issues/16323 (segfault, which I do not know if what is happening when it crashes on Linux, I don't get a chance to see any errors and I am not sure where the log is)
https://github.com/FreeCAD/FreeCAD/issues/17596 (cannot fillet an outside edge which looks like it should be perfectly fine to do)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.2.0
  * 3DfindIT 1.2.0
  * Help 1.0.3
  * FEM_FrontISTR 0.1.0
  * 3D_Printing_Tools
```

### Subproject(s) affected?

PartDesign

### Anything else?

[28mm spool with trough and cooling holes v1-2.zip](https://github.com/user-attachments/files/19140042/28mm.spool.with.trough.and.cooling.holes.v1-2.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-08:
You have combined the tools from the PartDesign Workbench with those from the Part Workbench without a full understanding of the ramifications. As a beginner you should avoid this. You have create a Part boolean from a PD body and Part objects. To fillet edges of the resulting object you should use the Part Fillet tool, not the PartDesign Fillet tool.

It is better to start a forum topic about such a problem before creating an issue on GitHub. I am closing this as a user error.

closed 2025-03-08 by Roy-043

Jimmeee, 2025-03-09:
Thank you for the explanation, that certainly could be why one of the situations I described was going wrong. However, let me point out a couple things:

1) IMHO, nothing should cause the program to just crash, so isn't this still a bug?
2) I tried doing the fillets with the Part tool instead of Part Design tool, and if the fillets were small enough it maybe worked (in some places it did, in other places, I did not see any difference), otherwise I got "Fillet: BRep_API: command not done" or the program crashed. 

In #2, I understand that the first error may be due to trying to add fillets larger than possible (large than the length of the side). At least it has been my observation that you cannot add a fillet larger than the side (which makes sense, I think).

In that case, a meaningful error message would be very useful (e.g., why was the command not done?). 

But again, in the case of a crash, the program should never just close, should it? Even if I screw up, I don't think that's intended behavior. 

So again, thank you very much for pointing out that there is a difference between these two tools, and for the very quick response. Maybe I can get it working now by starting from scratch (because my files, no matter which tool I use, don't seem to work). But, the complete crashes and lack of meaningful error messages seem like issues to me. 

By the way, I did  uninstall 3DfindIT 1.2.0 and 3D_Printing_Tools just because they were not necessary, in case they were doing anything strange. It did not change anything.

Roy-043, 2025-03-10:
It is indeed the case that not all error message are meaningful. This is partly due to the OCC kernel that FreeCAD relies on. FreeCAD is often just passing on whatever message it receives from the kernel.

If you have a reproducible scenario for a crash you should create a separate GitHub issue. I am leaving this issue closed as its main subject has been addressed.

I would again like to suggest the forum as the 1st port of call when encountering any FreeCAD issue.