FreeCAD/FreeCAD #20065: Probs with the BBS and github from the help menu

maxlittlemore, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Has anyone made a new email address and tried to join the BBS? Without internal knowledge? Fail every time.

Can't retry for (some arbitrary amount) seconds.....
(actually it doesn't report seconds, just says you can re try for 'a time')

How do we get help?

to use Australian English, youz kunts are fucked. 

Fork and start again, guess it's gone the way of Gnome

All the best,

### Full version info

```shell
version 1.0

Please make the question clearer. Again, assume I'm idiot with no understanding of how huge your member must be because of the code you wrote....

The text here is ambiguous. Seems to be a pattern. Happy to help.
```

### Subproject(s) affected?

None

### Anything else?

please spent at least some time checking the world-facing parts...

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-08:
What are you trying to do?

maxlittlemore, 2025-03-08:
join the BBS to get help, register, get help.
What I'm actually trying to do is have greedy selection on defined edges and do a chamfer on them... not possible in some odd edit mode I find myself in, try to get help online and I find a world of pain, more painful than dealing with the KUNTS of GTK.

maxwxyz, 2025-03-08:
You can try to get help on the forums: https://forum.freecad.org/ or Discord

closed 2025-03-08 by maxwxyz

maxlittlemore, 2025-03-08:
thanks. not linked from help in the menu

I'm gunna slice vomit any moment now... (does github understand aussie swearing?

maxlittlemore, 2025-03-08:
or give my diets to discord> is this a thing?