FreeCAD/FreeCAD #20068: FreeCAD 1.1.0, Libs: 1.1.0devR40586 (Git) won't open on macos

looooo, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I cannot open freecad 1.1 on macos arm anymore. I tried to remove the FreeCAD cache but the issue still appears:
```
FreeCAD 1.1.0, Libs: 1.1.0devR40586 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Error in InitApplications lstat: embedded null character in path
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 707, in <module>
  File "<string>", line 98, in InitApplications
  File "<frozen posixpath>", line 445, in realpath
ValueError: lstat: embedded null character in path
--------------------------------------------------------------------------------
(qt.qpa.fonts) Populating font family aliases took 57 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.
Traceback (most recent call last):
  File "<string>", line 303, in <module>
  File "<string>", line 133, in InitApplications
<class 'AttributeError'>: module 'FreeCAD' has no attribute '__ModDirs__'
Error in FreeCADGuiInit.py: module 'FreeCAD' has no attribute '__ModDirs__'
```

### Full version info

```shell
>>> FreeCAD.Version()
['1', '1', '0', '40586 (Git)', 'https://github.com/FreeCAD/FreeCAD main', '2025/03/07 15:26:29', 'main', '187630077f89733c8b6d0a74554916a0975e5146']

installed with pixi global.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dogunter, 2025-03-12:
I can confirm the latest packaged weekly also fails in the same way.

```
% /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Error: Failed to open library "/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib"! 
Error: dlopen(/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib, 0x0005): tried: '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file, not in dyld cache)!
FreeCAD 1.1.0, Libs: 1.1.0devR40655 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Error in InitApplications embedded null byte
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 707, in <module>
  File "<string>", line 98, in InitApplications
  File "<frozen posixpath>", line 416, in realpath
  File "<frozen posixpath>", line 451, in _joinrealpath
ValueError: embedded null byte
--------------------------------------------------------------------------------
(qt.qpa.fonts) Populating font family aliases took 197 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
Traceback (most recent call last):
  File "<string>", line 303, in <module>
  File "<string>", line 133, in InitApplications
<class 'AttributeError'>: module 'FreeCAD' has no attribute '__ModDirs__'
Error in FreeCADGuiInit.py: module 'FreeCAD' has no attribute '__ModDirs__'
2025-03-12 09:02:49.533 freecad[60291:26595660] +[IMKClient subclass]: chose IMKClient_Modern
2025-03-12 09:02:49.533 freecad[60291:26595660] +[IMKInputSession subclass]: chose IMKInputSession_Modern
```

rtm7777, 2025-03-15:
I've got the same error with 40655 build

yuri-karadzhov, 2025-03-16:
Same with intel version (40655).

thomasgi1, 2025-03-16:
Perhaps that code in `src/App/Application.cpp` no longer works on MacOS since commit 8ff4b1b018 causing the embedded null character in path:
```
std::string Application::getLibraryDir()
{
#ifdef LIBRARYDIR
    // #6892: Conda may inject null characters => remove them
    auto path = std::string(LIBRARYDIR);
    QDir dir(QString::fromStdString(path));
    if (dir.isAbsolute())
        return path;
    return mConfig["AppHomePath"] + path;
#else
    return mConfig["AppHomePath"] + "lib";
#endif
}
```

thomasgi1, 2025-03-16:
See #6892 for details

closed 2025-03-17 by hyarion