FreeCAD/FreeCAD #20071: PartDesign: Fillets and chamfers are automatically propagated to edges tangent to selection

pierreporte, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Say you want to make a fillet on a part, along a straight edge with fillets on each side, but only along this edge:

[image]

After activanting `PartDesign_Chamfer`, you select the edge:

[image]

You will get this:

[image]

It is exactly the same with fillets. These features don’t allow the user to set the prolongation of selection. It is always automatically propagated in tangency. FreeCAD lacks an option to either select tangency propagation or minimal propagation. Tangency should remain the default.

The workaround for simple cases is making pockets or sweeps along a path.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell ..7.0, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.0236/117.596
Installed mods: 
  * CosmeticThread3D
  * sheetmetal.backup1731333427.8204525 0.5.3 (Disabled)
  * Curves 0.6.59
  * CurvedShapes 1.0.13
  * sheetmetal 0.7.21
```

### Subproject(s) affected?

PartDesign

### Anything else?

Catia provides the option:

[image]

Two propagation modes are given: tangency (default) or minimal.

From the [documentation](http://catiadoc.free.fr/online/prtug_C2/prtugbt0602.htm):

> **Propagation**
>
> Two propagation modes are available: 
>
> * **Minimal**: edges tangent to selected edges can be taken into account to some extent. The application continues chamfering beyond the selected edge whenever it cannot do otherwise. In our example below, the chamfer is computed on the selected edge and on a portion of tangent edges:
>    [image]
> * **Tangency**: the application chamfers the entire selected edge as well as its tangent edges. It continues chamfering beyond the selected edge until it encounters an edge that is non-continuous in tangency as shown in our example:
>   [image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-08:
You can also use the fillet or chamfer tool in Part WB which provides an option to not follow tangent edges, only use selected edges and even different radii for individual edges.

Syres916, 2025-03-08:
> You can also use the fillet or chamfer tool in Part WB which provides an option to not follow tangent edges, only use selected edges and even different radii for individual edges.

I know about the variable radius but I don't remember seeing the option to not follow tangent edges, is the Wiki out of date https://wiki.freecad.org/Part_Fillet or is it a power users only parameter, screenshot required?

maxwxyz, 2025-03-08:
You are correct, I might have remembered it wrong or maybe it has worked in the RT branch. It also propagates automatically in Part. I'll add the Part WB label too and hide the comments above.

hyarion, 2025-03-11:
I checked and:
1. RT's branch does _not_ have a fix for this
2. I don't find an easy way in OCCT to specify that it should _not_ follow tangency edges.

Adding an option, similar to radii, to specify how to handle propagation should might be difficult in FreeCAD without changing OCCT. Adding this to OCCT seem doable though.

My (limited) understanding of OCCT is that to change this to allow configuring how the propagation works, we would have to change `BRepFilletAPI_MakeFillet`/`BRepFilletAPI_MakeChamfer`, `ChFi3d_FilBuilder`/`ChFi3d_ChBuilder`, and possibly some of the `ChFiDS_Spine` classes.

@blobfish you seem to have quite some knowledge on OCCT, do you have any thoughts on the matter or recommendations?

pierreporte, 2025-03-11:
@pieterhijma You could be interested in this since you’re now in charge of discussing with OpenCASCADE.

pieterhijma, 2025-03-14:
> [@pieterhijma](https://github.com/pieterhijma) You could be interested in this since you’re now in charge of discussing with OpenCASCADE.

Thanks, added it to my project.

cmoliverio, 2025-09-18:
Hey folks,

I just want to verify that this issue is still open.  @pieterhijma was this capability added to OCCT?  I would love to help.

pierreporte, 2025-09-18:
Yes the issue is open.

pieterhijma, 2025-09-18:
I don't think OCCT supports that. As far as I can tell, fillets and chamfers are always propagated in tangency in OCCT. In the example above, it is quite easy to do that, namely first do the chamfer and then the fillets. So, if we want this feature, then we would have to submit a feature request to OCCT. If this particular feature is important to us, it would be good to come up with a better example that does not allow you to solve this by simply reordering the operations. @pierreporte, do you happen to have ideas for that?

codeotaku, 2026-09-01:
@fredvol  - this may be possible with the [upcoming fillet update(s)](https://github.com/FreeCAD/FreeCAD/issues/5723)

codeotaku, 2026-09-01:
Progress on this:

[image]

[image]

[image]

[image]

[image]

[image]

Chamfers: 
[image]

[image]

[image]

[image]

[image]

[image]

maxwxyz, 2026-09-01:
Checkbox maybe?

fredvol, 2026-09-01:
Look like a super nice upgrade , thanks for your work! 
Looking forward to test that.

codeotaku, 2026-09-01:
@maxwxyz I think checkbox and a tolerance/angle threshold (maybe the threshold will only make sense after we get the occ hardcoded tolerance value updated, though).