FreeCAD/FreeCAD #20075: in TechdrawCircleCenter lines created in v1.0.0 do not cross in the center of the circle

JuanP74, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

the CircleCenter lines created in v1.0.0 do not cross in the center of the circle as in the standard iso 128-24. see pictures in this link:
https://discord.com/channels/870877411049357352/1259157480982712410/1347916505391566899

### Full version info

```shell
OS: macOS High Sierra (10.13)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.32
  * kicadStepUpMod 11.5.5
  * slic3r-tools
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pierreporte, 2025-03-08:
Images from the discord for those who aren’t on it:

[image]

[image]

Roy-043, 2025-03-08:
If I understand correctly this is a linetype generation issue. Instead of starting at an endpoint, generation should start from the center of the line so that there is a dash there.

WandererFan, 2025-03-09:
I know this is the traditional way of drawing intersections of non-continuous lines, but my copy of ISO128-24:2014 does not mention line intersections at all.   

All the illustrations in my copy use this tradition though.

JuanP74, 2025-03-09:
the specific requirement is given in ISO 128-20:
5.2.2 Representation
The requirement of 5.2.1 shall be fulfilled by starting the lines at the junction (see figure 14) or by using a complete,
or partial, cross formed by dashes

pierreporte, 2025-03-09:
@JuanP74 thanks! Does it say anything about patterns?

maxwxyz, 2025-03-10:
@pierreporte it does not work for patterns, here something from an iso 

[image]

WandererFan, 2025-03-10:
from: ISO 128-20:1994

5.2.1 The basic types of lines, Nos. 02 to 06 and Nos. 08 to 15 should **preferably** meet at a dash;

in 5.2.2, the recommended approach will only work on line styles where the first element is NOT a gap.  Dashed and Double Dash Dotted both start with a gap.

JuanP74, 2025-03-10:
> [@JuanP74](https://github.com/JuanP74) thanks! Does it say anything about patterns?

I think the standard has been revamped a bit into ISO 128-2, I guess the latest issue is this one (https://www.iso.org/standard/83355.html) but I do not have a copy, you can see a preview here: https://cdn.standards.iteh.ai/samples/83355/3b66078cc4f2436b9738b6a0a86f10ce/ISO-128-2-2022.pdf
Looks like there is an appendix dedicated to implementing it on CAD (I guess it's the content of former ISO 120-21 now withdrawn, ISO 128-20 seems withdrawn as well).
Current status of ISO 128 is described here: https://cdn.standards.iteh.ai/samples/65296/4c94466a19244bf59167c2129987e884/ISO-128-1-2020.pdf

pierreporte, 2025-03-10:
The FPA could provide access to the standard if needed. They already bought access to some of two of them.

Ping @chennes

WandererFan, 2025-03-11:
The FPA has these in its library:
ASME Y14_3_2012_R2018 - Orthographic and pictorial views
ISO 128-3_2022 - Technical product documentation part 3

Issue for purchase of 128-2: https://github.com/FreeCAD/FPA/issues/316

FEA-eng, 2026-10-18:
From what I know, it wasn't fixed.