FreeCAD/FreeCAD #20076: Part / PartDesign ignore the common preference for colors

maxwxyz, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Despite only having one place to set the default color, the same sketch extruded in Part WB has a white line and vertex color (top object), padded in Part Design the line and vertex color is black (bottom body, correct as set in the preferences).

[image]

This is the common preference page to set default color and line + vertex color for Part and Part Design:

[image]

File:

[PD-Part.FCStd.zip](https://github.com/user-attachments/files/19145185/PD-Part.FCStd.zip)

This bug is also present in safe mode and in 1.0 with safe mode.

@kadet1090 FYI, maybe related to auto sketch color?
@MisterMakerNL FYI regarding stylesheets

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.21
  * Silk 0.1.5
  * Telemetry 1.0.0
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-08:
I cannot reproduce in light theme which is even more absurd. @kadet1090 I think it is because of the auto sketch color which affects the resulting solid which is based on a sketch, but only in Part. In Part Design the Body/feature is always with black lines as set in the preference correctly.

[image]

Syres916, 2025-03-09:
The code in Part is to use the point/line colour for primitives (black in Safe Mode) whether that be cube/helix etc but when it's creating a parent based on another object it's been hardcoded for years to inherit the line/point colour from that existing object so a Sketch would be white by default but if the user had a Draft>Rectangle that was the basis for a Revolve then they would be black by default.

[image]

maxwxyz, 2025-03-09:
understood, but the sketch auto-color was newly added which results in mixed colored solids

PaddleStroke, 2025-09-05:
My suggestion is that Part should detect if the selection is a sketch, and if so it should use the default color, not the sketch color.

closed 2025-09-05 by kadet1090