FreeCAD/FreeCAD #20078: Stray features outside a body

YosypC, 2025-03-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

As from the image, from the **Thickness** to the **Hole** features are outside a body. I have no idea at which step they have gotten out of the base **Body**.

As far as I know, FreeCAD Part Design WB's features need to stay inside one. Initially, I moved **Pad** inside myself, otherwise it was completely stray. The other features yield a _"Some of the selected features have dependencies in the source body"_ after attempting a _"Move object to another body"_.

All of this behaviour has the consequence that at any visibility change of a single feature, the others' are not affected. Along other possible implications unbeknownst to me.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Anything else?

Rename file extension in FC's: [ReggimensolaWTF.zip](https://github.com/user-attachments/files/19146287/ReggimensolaWTF.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
Can you add steps how those features get out of the body? Did you delete something?

YosypC, 2025-03-09:
I honestly did not realize until much later. I remember having created a new active body, then creating a subshape binder of the Pad, and some sketches. I did not do much, but unfortunately I can't remember the precise point where it happened. I might be able to reproduce by creating a new file, since this one appears to be toast.

FEA-eng, 2025-10-01:
@YosypC Do you now know exactly what actions caused it? We need them to reproduce (and potentially fix) the issue.

YosypC, 2025-10-01:
@FEA-eng I was very surprised to come across this problem since I didn't veer off my usual workflow. I have absolutely no idea what could have caused the issue. When I noticed, the history was also gone so I couldn't trace it back.

FEA-eng, 2026-10-19:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```