FreeCAD/FreeCAD #20081: The Trim tool sometimes trims lines incorrectly.

kaiwas, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The problem is very rare, but it still exists. Sometimes it happens 1 out of 20 times. Sometimes almost immediately.
I don't know what it is connected with. Often undoing the action and then repeating it solves the problem.

Video demonstration. 

https://github.com/user-attachments/assets/2990a5f4-87f0-4819-881f-7065eb3d233e

Just do this a few times and sooner or later you will encounter it.
It feels like the trim works twice with a single click.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Logical/physical DPI: 96/91.7938
Installed mods: 
  * Test
  * PitchedRoof
  * addFC.backup1741509994.3771434 (Disabled)
  * Road 2025.2.22
  * addFC
  * Curves 0.6.59
  * freecad.gears 1.3.0
  * sheetmetal 0.7.21
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
could you add an example file?
@AjinkyaDahale FYI

kaiwas, 2025-03-09:
> could you add an example file?

the file won't help. a repeated operation in the same place does not reproduce the problem.
to translate it is enough to do the same operation as in the video. if you are lucky - the problem will appear on the first step. if you are unlucky - on the "twentieth".
in any case, I manage to reproduce it after several attempts

pure-machinery, 2025-03-09:
I previously had similar issues with the trimming tool, although my geometry was generated by a macro - which shouldn't impact the trimming logic. If i recall correctly it showed the correct points between which it would trim to, but once trimmed it would trim another part of the curve entirely.

AjinkyaDahale, 2025-03-11:
I think what's happening is accidentally the "continuous trim" is being triggered because the mouse moves slightly before preselection is refreshed. I'm curious if it happens only in development and not in 1.0, since that part of the code is not yet touched to refactor.

kaiwas, 2025-03-11:
> I'm curious if it happens only in development and not in 1.0

I checked in the stable version 1.0
This problem also appears there.

I don't want to try 0.21, so as not to spoil the configuration files.

macdroid53, 2025-04-03:
Open a sketch.
Create two over lapping rectangles.
Attempt to Trim all overlapping lines.

Sometimes it removes the appropriate edge, sometimes not. Sometimes FC crashes.

Even when it works, it throws errors:
`08:23:53  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: Unknown C++ exception
08:23:53  Unnamed#Sketch: App.getDocument('Unnamed').getObject('Sketch').trim(7,App.Vector(27.902954,-7.019064,0))
08:23:53  Unnamed#Sketch: Failed to trim edge
`

On this build/platform it works (with the aforementioned error):
`OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41135 (Git)
Build date: 2025/04/03 03:28:51
Build type: Release
Branch: main
Hash: 72c1659e4dc91a9d09fa688b8173f5a3e0d57531
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/93.6146/1
Installed mods: 
  * Silk 0.1.5
  * Defeaturing 1.2.3
  * Curves 0.6.60
  * CurvedShapes 1.0.13
  * BlenderLiveLink
  * DynamicData 2.73.0
  * lattice2 1.0.0
  * sheetmetal 0.7.22
  * Manipulator 1.5.7
  * kicadStepUpMod 11.5.7
  * fasteners 0.5.37
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.3.0
  * ExtremeProDark 2.9.0
`

On this build/platform it works on the first two or three edges, then indicates the edge, but will not trim, also throwing aforementioned error:
`OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41135 (Git)
Build date: 2025/04/03 03:28:51
Build type: Release
Branch: main
Hash: 72c1659e4dc91a9d09fa688b8173f5a3e0d57531
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/103.66/1
Installed mods: 
  * lattice2 1.0.0
  * Curves 0.6.60
  * ArchTextures
  * Nodes 0.1.36
  * OpenTheme 2024.9.1
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * BlenderLiveLink
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * Silk 0.1.5
  * kicadStepUpMod 11.5.7
  * ThreadProfile 1.93.0
  * DynamicData 2.73.0
  * workfeature-macro
  * fasteners 0.5.37
  * Defeaturing 1.2.3
  * CurvedShapes 1.0.13
`

AjinkyaDahale, 2025-04-03:
@macdroid53 #20150 is meant to fix this issue. If you can compile please confirm.

macdroid53, 2025-04-06:
> [@macdroid53](https://github.com/macdroid53) [#20150](https://github.com/FreeCAD/FreeCAD/pull/20150) is meant to fix this issue. If you can compile please confirm.

See my comments in this issue: #810875 different result on different platforms for build 41155

AjinkyaDahale, 2025-04-14:
@kaiwas, is this still an issue?

kaiwas, 2025-04-14:
I forgot to check.

I made this figure three times. The problem does not reproduce. Everything works great!

[image]

kaiwas, 2025-10-01:
problem has been solved.