FreeCAD/FreeCAD #20083: sketcher_translate produces many new DOF

marvin42b, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using the Move/Array transform (W) in sketcher the resulting sketch has many new DoF (degrees of freedom) namely both x- and y-coordinates for all new copies.
It would be very helpful to automatically add point-coincident-constraints from the original to the copies.

Two cases:
(A) both start and endpoints of the translation(s) belong to the selected set of elements:
These two or three points are used for the coincidences between parent and child.
(B) otherwise:
The set of elements has a "first" point. This should be used together with a blue construction geometry vector to fix the children.

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/oxygen
Installed mods: 
  * fasteners 0.5.24
  * ThreadProfile 1.90.0
  * 3D_Printing_Tools
```

### Subproject(s) affected?

None

### Anything else?

I vaguely remember that this suggestion already was the default in some former versions of FreeCAD.

Side-note not worth a separate issue: I find it quite irritating that for the columns (copies) the number is 0-based counting the number of *additional* elements while for the rows it is 1-based counting *the total number* of rows.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-03-09:
this suggestion looks good, but it is practically useless, because the mentioned tools completely destroy previously set up constraints. you will still have to do a lot manually.

in my personal opinion, the operations of moving, arrays, rotations and scale in the current implementation are completely useless.
better to make a copy-paste

marvin42b, 2025-03-09:
Moving (and duplicating, I assume) seems to work O.K. in your case. 
What you point out is for rotation and scaling. 
But, frankly, for a *rotation* it is hard to maintain *vertical* and *horizontal* constraints 🤔
For scaling, you're right. I don't see the need for dropping the latter, but then of course the lengths *need* to be changed by definition.

closed 2026-10-31 by maxwxyz