FreeCAD/FreeCAD #20085: Freeze on filet creation

mgth, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I made a rounded cube, made a square hole on a face. Then when I try to create a filet at the for hole edges Freecad freezes. 
The bug is present on V 1.0.0

Test file : 

[RoundedCude Filets freeze.zip](https://github.com/user-attachments/files/19150130/RoundedCude.Filets.freeze.zip)

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/79.7968
Installed mods: 
  * CurvesWB 0.6.12
  * Defeaturing 1.2.3
  * ExtremeProDark 2.6.4
  * fasteners 0.5.37
  * FreeCAD-Ribbon 1.7.3.10
  * FreeCAD-themes 2025.1.7
  * freecad-xr-workbench 0.9.4
  * freecad_streamdeck_addon 0.1.7 (Disabled)
  * OpenDark 2024.3.13
  * OpenTheme 2024.9.1
  * parts_library
  * parts_library.bak
  * ProDarkThemePreferencePack 1.0.0
  * SearchBar 1.4.1
  * workfeature
```

### Subproject(s) affected?

None

### Anything else?

If I select a two or tree edges, it won't crash, but I get buggy filets. Note that trying to reproduce from start, I got different behaviors, but always wrong.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

mgth, 2025-03-09:
Note that chamfer don't freeze, but is still buggy. Especially not that one corner has reversed curve.
The reversed curve also occurs if you create a filet using the four inner faces.

[image]

[image]

Roy-043, 2025-03-11:
Fillets are a known weakness of the OCC kernel that FreeCAD relies on. You have to use a workaround here. If you want to fillet/chamfer the internal edges of the pocket, you should base the pocket on a filleted/chamfered sketch. If the pocket should stay 'square' you should base it on a sketch that is a tiny bit bigger than the face you are using now. Both workarounds avoid that too many edges meet at the same point (where the sperical corners touch the pocket).

Perlovka, 2025-04-03:
It freezes for about 30 seconds even on simplest shapes. Trying to export this to **3mf** or **stl** leading to hang forever.

[image]

Perlovka, 2025-08-29:
Any workaround? FreeCAD becomes completely unusable when it freezes every minute 🙅

Perlovka, 2025-08-29:
I can't believe. In my case problem is solved by just removing config directory 😆
Spent whole day compiling different versions of OCC 🤦

pieterhijma, 2026-09-29:
Reported in https://github.com/Open-Cascade-SAS/OCCT/issues/1571 with DRAW script [faulty-shape-fillet-rounded-cube.zip](https://github.com/user-attachments/files/32809525/faulty-shape-fillet-rounded-cube.zip).