FreeCAD/FreeCAD #20086: subshapebinder fails to add some line portions when adding multiple line loops from a sketch

paulgcrabb, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Some line segments are not being transferred to a subshapebinder from a sketch with multiple line loops

https://github.com/user-attachments/assets/e30a6d76-1106-4e8d-9975-c03c17d23def

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/93.2987
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
Wow I did not know that it works this way too.
- When you select the edges in the 3D view (no sketch active) and drag it to the binder, does it work?
- When you select all edges and then click on the SubShapeBinder icon in the toolbar to create the binder from the selection, does this work?

Could you add your example file?

paulgcrabb, 2025-03-09:
> * When you select the edges in the 3D view (no sketch active) and drag it to the binder, does it work?
Yes, it works, you notice that the Sketch gets highlighted in blue automatically.

> * When you select all edges and then click on the SubShapeBinder icon in the toolbar to create the binder from the selection, does this work?
Not if you are in a sketch because the subshapebinder icon is not available while in a sketch, however, create an empty subshapebinder in the body ( could be a different body ) and make the line selections you want in the sketch, while still in the sketch drag the sketch onto the new subshapebinder to transfer the line loops
> 
> Could you add your example file?
It's a really simple thing to recreate !

luzpaz, 2025-03-13:
>  It's a really simple thing to recreate !

@paulgcrabb we have lots of tickets to address. Please create an example file, zip it and drag/drop it into the Github textbox so testers can quickly test this bug. TIA.

paulgcrabb, 2025-03-14:
> > It's a really simple thing to recreate !
> 
> [@paulgcrabb](https://github.com/paulgcrabb) we have lots of tickets to address. Please create an example file, zip it and drag/drop it into the Github textbox so testers can quickly test this bug. TIA.

Done - at the top of the original post.

FEA-eng, 2026-10-25:
[image]

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

paulgcrabb, 2026-10-25:
It's 18 months since I started trying to get some sensible usable "extrude" functionality in Freecad... it's still un-usable so I have given up all hope for Freecad being anything other than a plaything for those who develop it - I don't think there is any interest in making the very core functionality actually fit for purpose.

Solidcraft has built more basic usable fucntionality for users in 4 weeks than Freecad has since it began...

maxwxyz, 2026-10-25:
have fun