FreeCAD/FreeCAD #20087: Cannot edit the PAD created from a subshapebinder

paulgcrabb, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I have multiple line loops in a subshapebinder and I created a PAD from a single line loop, I cannot subsequently edit the PAD and include additional line loops as part of the PAD. I can use multiple line loops when creating a new PAD from a subshapebinder.

I tried to modify it from the "data" tab associated with the PAD but it is not at all obvious how to control it from there and it's a very poor substitute for simply being able to select/deselect exisiting line loops/areas from the 3D interface.

https://github.com/user-attachments/assets/1b5c0148-c5f4-4452-b430-96bd152fc753

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/93.2987
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
You have to enable multiple solids in the body, this is still experimental

maxwxyz, 2025-03-09:
By the way: you can also just select the circle from the sketch and directly pad the selected circle, there is no need for an extra subshapebinder. Just note that the selected profile(s) must be closed shapes for a pad/pocket feature.

closed 2025-03-09 by maxwxyz

paulgcrabb, 2025-03-09:
"Allow Compound" was set to true, and "multiple bodies" was selected in "preferences"

maxwxyz, 2025-03-09:
Ah I have just noticed your videos have audio...
In the properties of the pad you can click on the three dots (...) in the `Profile` property and select another profile or clear the first:

[image]

paulgcrabb, 2025-03-09:
@maxwxyz Thanks for the quick discussion :)  - I have been playing with subshapebinders a lot over the last week or so, I know I can PAD directly from a sketch... but I can't do this if the sketch is part of a different body. I really like the possibilities that subshapebinders offer

maxwxyz, 2025-03-09:
Click on the three dots and Ctrl+Select the other profile.

paulgcrabb, 2025-03-09:
I tried to adjust it using the "profile" but I could not get it to work, probably my lack of knowledge about how to modify stuff in the "profile" section... my argument is that it should as easy to modify it as it was to initially select it when creating the PAD

I will try to understand the "Profile" properly

maxwxyz, 2025-03-09:
That is another issue :) https://github.com/FreeCAD/FreeCAD/issues/12721

paulgcrabb, 2025-03-09:
[image]

paulgcrabb, 2025-03-09:
Was this ticket closed as it is not considered a bug ? - I don't mind , just trying to understand the reason

paulgcrabb, 2025-03-09:
@maxwxyz - worked out how to modify the elements in the "profile" - many thanks for your input on this !