FreeCAD/FreeCAD #20089: Regression: Attachment Support/Linked Object selection dialog not working

jmickelin, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Several different functions in FreeCAD use a dialog simply titled "Link" for selecting an object in the project. Two prominent examples are for selecting values for the `Attachment Support` property (e.g. for sketches) or modifying the target of a Link by editing the `Linked Object` property. There seems to be a regression making you unable to use this dialog to actually make any changes.

In previous versions, this dialog stayed open as you selected or deselected objects shown in the list, and only closed when you hit OK. This is no longer the case. The dialog closes immediately when you click anything in the list, and inspecting the properties of the object afterwards reveals that no changes have taken effect.

I've attached a screen recording where I'm illustrating trying to use it for changing the support of a sketch.

I have confirmed that the properties in question aren't set to "Read-Only" by accident, so I would expect them to be editable.

### Full version info

```shell
I have tried this with both version 1.0.0 and the latest (as of 2025-03-09) 1.1 AppImage version available from https://github.com/FreeCAD/FreeCAD-Bundle/releases. The screen recording is from the former, but since that doesn't provide particularly specific version information in the About dialog, I'll post the one from the AppImage version too, for completeness.

--

# NixOS package:

OS: NixOS 24.11 (Vicuna) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.Unknown
Build type: Release
Python 3.11.11, Qt 5.15.16, Coin 4.0.3, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Manipulator 1.5.7
  * symbols_library
  * Help 1.0.3 (Disabled)
  * ExplodedAssembly
  * Assembly4 0.50.15
  * ThreadProfile 1.92.0
  * workfeature
  * MeshRemodel 1.10.34
  * Gridfinity 0.6.0
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.51
  * fasteners 0.5.32
  * Pyramids-and-Polyhedrons
  * OpenTheme 2024.5.16
  * CurvedShapes 1.0.13
  * workfeature-macro

--

# AppImage package:

OS: NixOS 24.11 (Vicuna) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/157.74
Installed mods: 
  * Manipulator 1.5.7
  * symbols_library
  * Help 1.0.3 (Disabled)
  * ExplodedAssembly
  * Assembly4 0.50.15
  * ThreadProfile 1.92.0
  * workfeature
  * MeshRemodel 1.10.34
  * Gridfinity 0.6.0
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.51
  * fasteners 0.5.32
  * Pyramids-and-Polyhedrons
  * OpenTheme 2024.5.16
  * CurvedShapes 1.0.13
  * workfeature-macro
```

### Subproject(s) affected?

Core

### Anything else?

https://github.com/user-attachments/assets/55db77ae-160e-40a2-be95-6b6dc4249d98

# Workarounds

For sketch supports, there is obviously the more common user-friendly way to change the geometry you're attaching to, by instead modifying the `Map Mode` property. This brings up the Attachment Editor.

For other properties, where you could previously use this dialog to make crude ad-hoc changes for dependencies within the project, there does not seem to be any user-friendly workaround exposed in the UI. However, I managed to make the changes I needed to a link by running an appropriate command in the Python console, like so:
```python
>>> App.getDocument('thumb_screws_handwheels_v2').getObjectsByLabel('Dimensions')[0].setLink(App.getDocument('thumb_screws_handwheels_v2').getObjectsByLabel('M8')[0])
```
This re-points the link labelled "Dimensions" to point to the object labelled "M8".

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
I cannot reproduce, could you try this in safe mode (Help>Restart in safe mode)?

jmickelin, 2025-03-09:
Interesting! Safe mode seems to have fixed it. I think I'd better set up a fresh profile and work on changing the settings to resemble what I currently have. If the issue does pop back up again, I'll update this issue :)

Thank you!

closed 2025-03-10 by maxwxyz