FreeCAD/FreeCAD #20091: Subshapebinder does not allow some enclosed areas to be selected for PAD/Pocket

paulgcrabb, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If multiple enclosed lineloops are added to a subshapebinder multiple areas are not available to be selected/highlighted when preparing to create a PAD ( or Pocket ), it is possible to generate a PAD for these unavailable areas by individually selecting all the line loops enclosing the area.

The ideal fix is to allow any enclosed areas to be selected by Click / CTRL+Click

https://github.com/user-attachments/assets/1c562e78-5f22-4657-9c0b-649cd3c80c5d

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/93.2987
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
Closely related to https://github.com/FreeCAD/FreeCAD/issues/12820

paulgcrabb, 2025-03-09:
@maxwxyz Thanks for taking note of these issues I have been having with subshapebinder, I read through https://github.com/FreeCAD/FreeCAD/issues/12820 and it looks like i have stumbled across almost working functionality and it appears that it is on the roadmap to be worked on.

My desire is to able to re-use parts of a sketch as much as is practical, especially if the ketch defines several bodies.

I know this is not really the place for a discussion but I think in conjunction with what has been discussed above extending the functionality of the PAD/Pocket tools so a PAD ( or pocket ) can "start from face" ( or even line or edge feature ) would bring Freecad very close to the key functionality that Fusion/Solidworks have.