FreeCAD/FreeCAD #20093: Sketcher - DOF highlighting doesn't work for PolyLine geometry

nv-steve, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

An under-constrained PolyLine will not have it's end vertices highlighted when the DOF link in the Sketcher Task panel is clicked.

In the below screen shot the top geometry was created from two discrete lines and a manually added coincident constraint. The lower one is a PolyLine.

[image]

Simple demonstration file attached.

[DOF highlighting not working on polyline.FCStd.tar.gz](https://github.com/user-attachments/files/19151194/DOF.highlighting.not.working.on.polyline.FCStd.tar.gz)

### Full version info

```shell
OS: Linux Mint 22 (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.40317 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b67ab979afdba808fc03196faf9c5f8830df471f
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96.0135/108.888
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

nv-steve, 2025-03-09:
Additional discussion on Discord suggests deleting constraints changes the highlighting behavior. In my testing I couldn't see this, but I may be doing it differently.

maxwxyz, 2025-03-09:
Does it appear correctly when moving the geometry? Then it would be this issue https://github.com/FreeCAD/FreeCAD/issues/19990

nv-steve, 2025-03-09:
It does. Good catch.

closed 2025-03-09 by maxwxyz