FreeCAD/FreeCAD #20094: The Addon Manager reports wrong version numbers, or does not report any version number at all

raindropsfromsky, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Here is a screenshot of AM after I updated 4 WBs.
[image]

There are four different problems:
1. Latest version not shown for the Curves WB.
2. The earlier version of Ribbon WB is shown as 0.0.0 (which is wrong)
3. The version number of SheetMetal WB has not changed at all.
    (Then why did it update?)
4. There are no version numbers shown for the versions_library.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical/physical DPI: 96/123.928
Installed mods: 
  * AnimationFreeCAD 1.0.0
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.59
  * ExplodedAssembly
  * fasteners 0.5.37
  * frame 0.1.1
  * FrameForge 0.1.2
  * FreeCAD-Ribbon 1.7.3.4
  * freecad.gears 1.3.0
  * FreeGrid 2.2.0
  * OpenTheme 2024.9.1
  * parts_library
  * Quetzal 1.3.3
  * SearchBar 1.4.1
  * sheetmetal 0.7.21
  * Silk 0.1.5
  * symbols_library
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-09:
@chennes is this something in the addon manager or does the addons add capabilities to show it there?

Syres916, 2025-03-09:
No 3 is because the Addon author doesn't always (really they should) update the version number even though there has been a new commit so that's not a core issue the AddonManager is doing it's best with the info provided.