FreeCAD/FreeCAD #20096: Auto-Change of element not triggering properly

drwho495, 2025-03-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[autohange-test-2.zip](https://github.com/user-attachments/files/19152718/autohange-test-2.zip)

This issue is a little hard to explain, but it is very annoying.

Whenever you update the 20mm dimension to 28mm in Sketch003, the sketch breaks.
What seems to be happening is an edge in Pocket007 is passing an edge in Pad005, and an auto change is not triggered. If you manually move the edge in Pad005 that is passing the edge in Pocket007, then an autochange IS triggered and the edge is setup. You can look at the video I made to try and help explain this bug.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical/physical DPI: 96/161.377
Installed mods: 
  * Render 2024.12.15 (Disabled)
  * ArchTextures
  * Assembly3 0.12.2
  * CadbaseLibrary 2.0.0
  * Behave-Dark-Colors 0.1.1
  * SearchBar 1.0.2
  * WebTools 1.0.0
  * Manipulator 1.5.7
  * POV-Ray-Rendering
  * cadquery_module 2.1.0
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * freecad.gears 1.3.0
  * ExplodedAssembly
  * fasteners 0.5.37
  * cadquery_module.backup1741030188.0603402 (Disabled)
  * FreeCAD-Telemetry 1.0.0beta
  * QuickMeasure 2022.10.28
  * FreeCAD-Ribbon 1.7.3.7
  * assembly2
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

PartDesign

### Anything else?

[video.zip](https://github.com/user-attachments/files/19152760/video.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-03-10:
This issue is not present in Link Branch 3

Syres916, 2025-03-10:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/14595 IMHO

drwho495, 2025-03-10:
Brad said that one was hard to fix; I believe this one is a bug where an autochange isn't triggered properly. If you move the edge that the first pad is surpassing manually, then it works properly.

maxwxyz, 2025-03-11:
Please add the info and files to #14595 it seems too similar

closed 2025-03-11 by maxwxyz

FEA-eng, 2026-02-03:
Still relevant. There are also errors after opening and recomputing the file.

```
09:12:25  <Sketch> SketchObject.cpp(9544): External geometry autohange_test_2#Sketch015.e1 missing reference: Pad002.;#f12:12;:U;XTR;:Hbc2:7,E
09:12:25  <Sketch> SketchObject.cpp(9544): External geometry autohange_test_2#Sketch015.e2 missing reference: Pad002.;#f12:11;:U;XTR;:Hbc2:7,E
09:12:25  <Sketch> SketchObject.cpp(9544): External geometry autohange_test_2#Sketch015.e3 missing reference: Pad002.;#f12:14;:U;XTR;:Hbc2:7,E
09:12:25  <Sketch> SketchObject.cpp(9544): External geometry autohange_test_2#Sketch015.e4 missing reference: Pad002.;#f12:13;:U;XTR;:Hbc2:7,E
09:12:25  <Sketch> SketchObject.cpp(9544): External geometry autohange_test_2#Sketch015.e9 missing reference: Pad002.;#f12:b;:U;XTR;:Hbc2:7,E
09:12:25  <Sketch> SketchObject.cpp(9544): External geometry autohange_test_2#Sketch016.e10 missing reference: Pocket.;#eef:23;:Hbb6,E
09:12:25  <Sketch> SketchObject.cpp(9544): External geometry autohange_test_2#Sketch016.e11 missing reference: Pocket.;#eef:25;:Hbb6,E
09:12:25  <PropertyLinks> PropertyLinks.cpp(518): autohange_test_2#DatumPoint.AttachmentSupport missing element reference autohange_test_2#Pocket007 ;#109a:6;:G2#128f;CUT;:H-bde:d,E.Edge77
09:12:25  AttachEngine3D::calculateAttachedPlacement: path curve second derivative is below 1e-14, cannot align X-axis.
09:12:25  PositionBySupport: AttachEngine3D::calculateAttachedPlacement: path has infinite radius of curvature at the point. Can't align for revolving.
09:12:25  AttachEngine3D::calculateAttachedPlacement: path curve second derivative is below 1e-14, cannot align X-axis.
09:12:25  AttachEngine3D::calculateAttachedPlacement: path curve second derivative is below 1e-14, cannot align X-axis.
09:12:25  <Exception> AttachEngine3D::calculateAttachedPlacement: path has infinite radius of curvature at the point. Can't align for revolving.
09:12:25  DatumPoint: AttachEngine3D::calculateAttachedPlacement: path has infinite radius of curvature at the point. Can't align for revolving.
09:12:25  AttachEngine3D::calculateAttachedPlacement: path curve second derivative is below 1e-14, cannot align X-axis.
09:12:25  <Exception> AttachEngine3D::calculateAttachedPlacement: path has infinite radius of curvature at the point. Can't align for revolving.
09:12:25  DatumPoint: AttachEngine3D::calculateAttachedPlacement: path has infinite radius of curvature at the point. Can't align for revolving.
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260127 (Git shallow)
Build date: 2026/01/27 15:33:30
Build type: Release
Branch: (HEAD detached at 7bcaa76)
Hash: 7bcaa766de101b10c03b4a6f4fd09292068d5ef6
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```