FreeCAD/FreeCAD #20098: Sketcher: Angle rendering broken for negative angles

PaddleStroke, 2025-03-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

If you use the arc tool for example. If the angle shown in the OVP is negative, then the rendering of the soDatumLabel is wrong. It was good before this is a regression.

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-10:
also related / fixed by https://github.com/FreeCAD/FreeCAD/pull/20101 ?

PaddleStroke, 2025-03-10:
I don't think so no

PaddleStroke, 2025-03-21:
unfortunately this issue can be reopened. It appears I have missed that the rendering is still not correct : 

[image]

- The arrows are in the wrong direction
- The hole in the middle of the arc is too big.

What it should look like : 

[image]

@FlachyJoe could you check it out please?

matthiasdanner, 2025-05-29:
The wrong arrows are not only on arcs but on every feature where an angle is displayed. For example lines 

[image]

I created a PR for that

PR #21645 (2025-06-02): Sketcher: Fix Arrow Direction of Angle Arrows

## Issues

fixes #20098
For negative angles the arrows on the end of the arc where rendered in the wrong direction. This PR fixes that. Positive Angles were correct before and were not changed
Also fixes the huge gap next to the angle box

## Before and After Images

Before:
* [image]
* [image]

After:
* <img width="293" alt="Screenshot
closed 2025-06-02 by hyarion