FreeCAD/FreeCAD #20106: BIM: can no longer add Arch Space boundaries

furgo16, 2025-03-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download test file and rename it removing the `.zip` extension: [TestSpace2.FCStd.zip](https://github.com/user-attachments/files/19162832/TestSpace2.FCStd.zip)
2. Open test file and switch to BIM workbench
3. Select a wall face that intersects the space
4. With the <kbd>Crtl</kbd> key pressed, select the Space object
5. Click on the "Arch Add" (plus sign) command button on the toolbar
6. **Expected**: the Space object is trimmed up to the location of the previously selected wall face (the boundary)
7. **Actual**: Python stacktrace:

```python
Running the Python command 'Arch_Add' failed:
Traceback (most recent call last):
  File "/snap/freecad/1327/usr/Mod/BIM/bimcommands/BimArchUtils.py", line 56, in Activated
    FreeCADGui.doCommand("Arch.addSpaceBoundaries( FreeCAD.ActiveDocument."+sel[-1].Name+", FreeCADGui.Selection.getSelectionEx() )")
  File "<string>", line 1, in <module>

module 'Arch' has no attribute 'addSpaceBoundaries'
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40582 (Git) Snap 1327
Build date: 2025/03/06 21:29:39
Build type: Release
Branch: main
Hash: aa90da1541648cdb433d622e06b2d9ad27c9d509
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.11
  * fasteners 0.5.37
  * fcVM
  * ezydxf
  * Arch
  * Assembly4 0.50.17
```

### Subproject(s) affected?

BIM

### Anything else?

This used to work at least until FreeCAD 0.21

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-03-10:
@yorikvanhavre is this a matter of just adding this to `src/Mod/BIM/Arch.py`?

```python
from ArchSpace import *
```

semhustej, 2025-03-10:
I can confirm the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

furgo16, 2025-03-12:
For those using FreeCAD 1.0, here's a workaround: add or remove boundaries via the Tasks panel, as described on https://wiki.freecad.org/Arch_Space#Usage