FreeCAD/FreeCAD #20111: Scaling of a sketch containing Bsplines does'nt work.

beppebizza, 2025-03-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Designing a foil profile I encounterd a bug when trying to scale a shape defined in a sketch.  The original shape is deformed illogically, while the preview of the final shape is prefect.  This happens only if BSpline are involved. 
I can bypass the bug by extruding the profile into a solid, than scaling the solid and recovering a wire from the edge.
best regards
Giuseppe Bizzarrini 
beppebizzar@gmail.com

[naca0012a-corrupted-after-scaling.zip](https://github.com/user-attachments/files/19167162/naca0012a-corrupted-after-scaling.zip)

[Bug in the sketcher app.zip](https://github.com/user-attachments/files/19167172/Bug.in.the.sketcher.app.zip)

### Full version info

```shell
Designing a foil profile I encounterd a bug when trying to scale a shape defined in a sketch.  The original shape is deformed illogically, while the preview of the final shape is prefect.  This happens only if BSpline are involved. 
I can bypass the bug by extruding the profile into a solid, than scaling the solid and recovering a wire from the edge.
best regards
Giuseppe Bizzarrini 
beppebizzar@gmail.com
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-10:
Please add full about info.
@AjinkyaDahale FYI

beppebizza, 2025-03-13:
[Bug in the sketcher app.pptx](https://github.com/user-attachments/files/19219580/Bug.in.the.sketcher.app.pptx)

beppebizza, 2025-03-13:
pptx contains another example of the bug

AjinkyaDahale, 2025-03-14:
Seems OP didn't need to close the issue? @maxwxyz @beppebizza?

AjinkyaDahale, 2025-03-14:
I see that likely the issue here is that the constraints are on knot points, but control points are free to move. Scale should be able to change without changing the shape, but for whatever reason there's a recompute that mixes things up.

beppebizza, 2025-03-16:
Please have a look to these files.  The control points weights chenges in scaling, thus deforming the spline.  IN this condition sometimes the sketch is still valid, sometimes the scketch becomes invalid.

beppebizza, 2025-03-16:
[Checkcontrolpointsweights.zip](https://github.com/user-attachments/files/19272546/Checkcontrolpointsweights.zip)

AjinkyaDahale, 2025-03-16:
I think if you keep (at least most of) the weights fixed, the issue of invalid sketches would go away. That gives too many variables for the solver to play with and sometimes causes issues. That said I haven't tested your file yet, so the issue may be different.

beppebizza, 2025-03-18:
> Oh wait I think I just understood your issue. The "control points" are
> treated as physical circles rather than physical representations of an
> abstract value. Right?

yes, correct.
Giuseppe Bizzarrini

commit 2032a9d8 (2025-05-08): Sketcher: Scaling of a sketch containing BSplines doesn't work (#20943)

* [FreeCAD-20111] Sketcher: Scaling of a sketch containing BSplines doesn't work
When we executing Scale cmd we don't will lose knots information from shape geometry and it cause the scaled spline build with issue knots and it lead solver gets compute issue.
1. Add string format function PythonConverter::makeArryString(const std::stringstream& ss) for processing the control pnts string, mults string and knots string.
2. Add mults and knots information for python obj to make BSplineCurvePy::buildFromPolesMultsKnots(Args... [... cut]
closed 2025-05-08 by kadet1090