FreeCAD/FreeCAD #20115: Sketcher: Editing references to external geometry

m-sundman, 2025-03-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sometimes a reference to external geometry is wrong, either because the external geometry changed or simply because a different external geometry is preferred. Deleting the reference and manually recreating it requires re-creating all constraints, too, which is error-prone and, frankly, frustrating.

Changing which external geometry a reference refers to can make editing existing designs an order of magnitude faster, even after the TNP fixes have been incorporated.

Realthunder's branch already has this feature, along with a host of closely related features, such as detaching a reference:
[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.8
  * CurvedShapes 1.0.4
  * ThreadProfile
  * fasteners 0.4.54
  * fcgear
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-11:
@PaddleStroke FYI

PaddleStroke, 2025-03-11:
We could ask RT to PR this?

realthunder, 2025-03-12:
I can give it a try maybe next week, but it's going to be a big PR.

m-sundman, 2025-05-09:
@realthunder Have you had a chance to take a look at this?
Btw, although I would still continue to support you on patreon, and I love your fork, it would be even nicer to get as many features as possible from your awesome fork merged into mainline. The closer they are the easier it is to cross-merge.

maxwxyz, 2025-06-16:
@PaddleStroke maybe you could take a look at this?