FreeCAD/FreeCAD #20116: Cutting with a cut causes a segmentation fault

magesorcerian, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Cutting with a cut causes a segmentation fault.

From a new file, create two objects and boolean cut one with the other.  Create a third object, and boolean cut with the result of the first cut.  Crash.

I've tried with two different builds of 1.1.0 as listed below.  Doesn't happen with 1.0.0
may be related to issue #19906 

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/137.612/1
Installed mods: 
  * fasteners 0.5.37
  * lattice2 1.0.0

OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40607 (Git)
Build date: 2025/03/08 01:10:58
Build type: Unknown
Branch: main
Hash: 50e80f4a03ac9eda45e2c3f80fde26d9121ad15e
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/137.612/1
Installed mods: 
  * fasteners 0.5.37
  * lattice2 1.0.0
```

### Subproject(s) affected?

Part

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-11:
Please add an example file. Does it work in the stable 1.0 and also in safe mode (Help>Restart in safe mode)?

ElementW, 2025-03-11:
@maxwxyz Reproducible on 1.1 weekly (2025/03/10) with [crash.FCStd.zip](https://github.com/user-attachments/files/19195858/crash.FCStd.zip). Just boolean cut the two toplevel objects.
<details>

```
OS: Void Linux (KDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

</details>

closed 2025-03-11 by chennes