FreeCAD/FreeCAD #20119: Solver failed to converge: unclear problem

YosypC, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

Whenever I try applying a coincident constraint to these points, the program throws a Python exception: 

```
11:19:45  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (3, 0)
11:19:45  PhoneHolder#Sketch: App.getDocument('Unnamed').getObject('Sketch').movePoint(3,0,App.Vector(0.474517,-0.979442,0),1)
11:19:45  ViewProviderSketch: Drag curve: Not able to move point with the id and type: (3, 0)
11:22:18  Sketch: Solving the sketch failed
```

I am completely unable to move anything in the sketch. The lines remain white with no indication where the problem is.

```
11:22:54  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (30, 0)
11:22:54  PhoneHolder#Sketch: App.getDocument('Unnamed').getObject('Sketch').movePoint(30,0,App.Vector(1.275925,7.420558,0),0)
11:22:54  ViewProviderSketch: Drag curve: Not able to move point with the id and type: (30, 0)
```

By Ctrl-Zing, the problem goes away.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Anything else?

Rename extension in FCStd: [PhoneHolder - Copy.zip](https://github.com/user-attachments/files/19180928/PhoneHolder.-.Copy.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-03-15:
Your action is problematic. You have a line and two arcs that have a tangent constraint with that line. Making the two marked points coincident would reduce the length of the line to zero. You would first need to delete the line and then (probably) apply a tangent constraint to the endpoints of the arcs

YosypC, 2025-03-15:
This is correct, however the program doesn't tell you in any way. It just says there's a problem, not where or how to fix it. You have to figure it out yourself, while all the lines are still white.

carribeiro, 2025-06-05:
I've had a similar issue with the same message while creating a B-spline. I've created the B-spline, and after finishing, the line is white and I can't move anything. 

In this case, I'm not using any constraints. I'm not sure if I have to fill a separate bug or if the root cause is the same (an inconsistency in the drawing, which makes it unmovable).

[image]

YosypC, 2025-12-09:
Yes, but the program doesn't throw tantrums anymore: applying a coincident constraint on those two points yields no error nor any log message: all the lines remain white and it lets me move anything normally as nothing happened.

https://github.com/user-attachments/assets/40348bbe-2bd4-44a4-9834-e8dcc7c65ba7

Shouldn't the user be notified of the failure to converge?

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0rc1.20251125 (Git shallow)
Build date: 2025/11/25 17:53:08
Build type: Release
Branch: (HEAD detached at 5376760)
Hash: 53767601ea65de7cd9f6590e7d5856afc0841635
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.201/1
```

FEA-eng, 2026-03-10:
[This](https://github.com/FreeCAD/FreeCAD/issues/20119#issuecomment-3355523105) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260303 (Git shallow)
Build date: 2026/03/03 22:57:35
Build type: Release
Branch: grafted,grafted
Hash: 637463ac354f093e14dacfc5059eb70c2b1216fd
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-09-09:
There's a linked PR, so it should still be relevant

closed 2026-09-12 by maxwxyz