FreeCAD/FreeCAD #20121: There is no way to attach a undocked document window back.

kaiwas, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Video demonstration.

https://github.com/user-attachments/assets/fc5ec5fe-1bf3-498d-8585-5258348442b2

The docked menu and hotkeys do not work.
Only restarting the application helps.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * PitchedRoof
  * addFC.backup1741673646.798512 (Disabled)
  * addFC.backup1741509994.3771434 (Disabled)
  * Road 2025.2.22
  * addFC 1.4.3
  * Curves 0.6.59
  * freecad.gears 1.3.0
  * sheetmetal 0.7.21
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-03-11:
It's quite rigorous, but there is an easy way to fix this problem.
You need to select some object. Then remove the selection.
Then switching works again.

https://github.com/user-attachments/assets/a8aee39c-788e-4231-a529-baba6b4a33ba

maxwxyz, 2025-03-11:
Can you reproduce in safe mode and also in 1.0?

kaiwas, 2025-03-11:
> Can you reproduce in safe mode and also in 1.0?

I can reproduce this in safe mode on another computer (1.1 dev).
I'll check version 1.0 later.

https://github.com/user-attachments/assets/fc17048f-5ae8-4ed8-a342-04d043e20f0a

semhustej, 2025-03-11:
I get different behavior on Windows. When I undock a window, it is placed bellow the main FreeCAD window. See behavior in the video:

https://github.com/user-attachments/assets/6d6559a7-d0ac-42de-9513-d6c4f5ef5624

I can reproduce with FreeCAD 1.0

It seems to me the problem is that when window is undocked, it looses focus. It looks like this manifests in different way on linux and windows.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

kaiwas, 2025-10-01:
I don't know at what point the changes were made, but the problem has been solved.