FreeCAD/FreeCAD #20123: Weekly build 40655 does not start up on macOS x86

depthoffocus, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Not sure if this should be reported here or in the FreeCAD-Bundle repo.

The current weekly build **40655** does not properly start up on macOS x86. 

I had removed the quarantine as normal with 

```sh
xattr -r -d com.apple.quarantine "/Applications/FreeCAD.app"
```

FreeCAD then verifies as normal on first launch but silently quits.

On subsequent launches the splash screen appears briefly then disappears. There is no crash. The app appears again to be silently quitting.

Starting the app in safe mode with:

```sh
open /Applications/FreeCAD.app --args --safe-mode 
```

does not help. 

(That is the correct way to start safe mode, verified with other builds.)

There is no console output at that point.

Information that might be useful:

MacBook Pro (13-inch, 2019, touchbar) -- `MacBookPro15,2`
macOS Sequoia 15.3.1 (24D70)

(I am unable to get the FreeCAD build info from the about panel because things do not get that far).

### Full version info

```shell
--
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-11:
Related https://github.com/FreeCAD/FreeCAD/issues/20068 ?

depthoffocus, 2025-03-11:
> Related [#20068](https://github.com/FreeCAD/FreeCAD/issues/20068) ?

It might be, yes! When I go in and try to run the FreeCAD binary directly, which would normally work, I get this output:

```sh
mike@Michaels-MacBook-Pro MacOS % cd /Applications/FreeCAD.app/Contents/MacOS
mike@Michaels-MacBook-Pro MacOS % ./FreeCAD                                  
FONTCONFIG_FILE=/etc/fonts/fonts.conf
FONTCONFIG_PATH=/etc/fonts
GIT_SSL_CAINFO=/Applications/FreeCAD.app/Contents/Resources/ssl/cacert.pem
HOME=/Users/mike
LANG=UTF-8
LD_LIBRARY_PATH=/Applications/FreeCAD.app/Contents/Resources/lib
LOGNAME=mike
OLDPWD=/Applications/FreeCAD.app/Contents/MacOS
PATH=/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin
PREFIX=/Applications/FreeCAD.app/Contents/Resources
PWD=/Applications/FreeCAD.app/Contents/MacOS
PYTHONHOME=/Applications/FreeCAD.app/Contents/Resources
PYTHONPATH=/Applications/FreeCAD.app/Contents/Resources
QT_MAC_WANTS_LAYER=1
SHELL=/bin/zsh
SHLVL=1
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.rMhAn1Bk25/Listeners
SSL_CERT_FILE=/Applications/FreeCAD.app/Contents/Resources/ssl/cacert.pem
TERM=xterm-256color
TERM_PROGRAM=Apple_Terminal
TERM_PROGRAM_VERSION=455
TERM_SESSION_ID=4F0E7279-A490-4C82-BCA6-F750568833A5
TMPDIR=/var/folders/hd/l4cpk90d7pb7ytrwzd2bfh880000gn/T/
USER=mike
XPC_FLAGS=0x0
XPC_SERVICE_NAME=0
_=/Applications/FreeCAD.app/Contents/MacOS/./FreeCAD
__CFBundleIdentifier=com.apple.Terminal
Running: /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Error: Failed to open library "/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib"! Error: dlopen(/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib, 0x0005): tried: '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file), '/System/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib' (no such file, not in dyld cache)!
FreeCAD 1.1.0, Libs: 1.1.0devR40655 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Error in InitApplications embedded null byte
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 707, in <module>
  File "<string>", line 98, in InitApplications
  File "<frozen posixpath>", line 416, in realpath
  File "<frozen posixpath>", line 451, in _joinrealpath
ValueError: embedded null byte
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 303, in <module>
  File "<string>", line 133, in InitApplications
<class 'AttributeError'>: module 'FreeCAD' has no attribute '__ModDirs__'
Error in FreeCADGuiInit.py: module 'FreeCAD' has no attribute '__ModDirs__'
2025-03-11 14:05:46.429 freecad[64733:18897337] +[IMKClient subclass]: chose IMKClient_Modern
2025-03-11 14:05:46.429 freecad[64733:18897337] +[IMKInputSession subclass]: chose IMKInputSession_Modern
```

depthoffocus, 2025-03-11:
Does the same with `--safe-mode`

I clearly do _not_ have this library on my system:

`/Library/Frameworks/3DconnexionNavlib.framework/3DconnexionNavlib`