FreeCAD/FreeCAD #20124: BIM_Materials Merge duplicates results in error

semhustej, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- create BIM material in a new project
- open `BIM_Material` window
- click `Merge duplicates` button
- error occurs

```

15:14:59  Traceback (most recent call last):
15:14:59    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40504-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimMaterial.py", line 214, in onMergeDupes
15:14:59      for om in mats:
15:14:59                ^^^^
15:14:59  NameError: name 'mats' is not defined. Did you mean: 'mat'?
```

I can reproduce the error in FreeCAD 1.0

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Anything else?

I tried changing mats to mat on line 214. The resulting error in this case is:
```

15:12:47  Traceback (most recent call last):
15:12:47    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40504-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimMaterial.py", line 214, in onMergeDupes
15:12:47      for om in mat:
15:12:47  TypeError: 'FeaturePython' object is not iterable
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-12:
A simple question, if there's only _one_ material in the dialog, surely the `Merge duplicates` button should be disabled as there's nothing to do??

semhustej, 2025-03-12:
> A simple question, if there's only _one_ material in the dialog, surely the `Merge duplicates` button should be disabled as there's nothing to do??

Yes, that would make sense. Not sure how much work that would be - I don't think it is really that important and users can figure that out.

The issue is valid with more then one material in the project.

yorikvanhavre, 2025-03-12:
This is really weird... In the current form, that code should never have worked (the name `mats` is not defined anywhere). But I used that feature already in the past 😅 

Ok just checked, this commit did it: https://github.com/yorikvanhavre/BIM_Workbench/commit/8b053de60311b8bfa8baba4876b65a720e0a7508  and I can't understand why I did it... 

I *think* the idea might have been that if you select several materials then you can "force" them to merge, even if their names don't match. But then it's easier to use the "Merge to..." context menu.

I'll remove that extra code, and add @Syres916 's suggestion.

commit 47d6ad84 (2025-03-26): BIM: Fixed material dialog's merge duplicates button - fixes #20124
closed 2025-03-26 by chennes