FreeCAD/FreeCAD #20126: Arch_Space $tag not showing anything

semhustej, 2025-03-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Steps to reproduce:**

- create Arch_Space object
- add $tag to the text field (according to wiki, $tag shows something https://wiki.freecad.org/Arch_Space#Properties)
- $tag is displayed as simple text

[image]

I am able to reproduce in FreeCAD 1.0

**Findings:**
Space has no Tag attribute in IFC schema, so it is correct for FreeCAD to create space without this attribute. 
https://identifier.buildingsmart.org/uri/buildingsmart/ifc/4.3/class/IfcSpace

**Proposed solution:**
Use $longname to show `Long Name` attribute instead of $tag.

With these options, user could separate room number to `Label` (Name) attribute and room name to `Long Name`. This is in accordance with IFC schema: _"LongName holds the full name of the space, it is often used in addition to the Name, if a number is assigned to the room, then the descriptive name is exchanged as LongName."_
https://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/IfcSpace.htm

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Anything else?

The issue was confirmed by several users in this forum thread: https://forum.freecad.org/viewtopic.php?t=95458

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-03-12:
We should actually extend the system. Ex: Anything that begins with $ is replaced with the contents of that property, if it exists. So $tag or $Tag would get the Tag property, $longname or LongName would be replaced by the Long Name property, etc...

semhustej, 2025-03-12:
That would certainly give the tool more flexibility, which would be nice.

yorikvanhavre, 2025-03-12:
Working on it ;)

PR #20207 (2025-03-17): BIM: Fixed space text display
Allow to use any of the Space object properties in the displayed text. Fixes #20126
commit d3aaa4cf (2025-03-17): BIM: Fixed space text display - fixes #20126
closed 2025-03-17 by chennes