FreeCAD/FreeCAD #20137: BIM: cannot create space if all boundaries belong to the same wall

furgo16, 2025-03-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:
 
2. Create a Draft rectangle. Make sure "Make face" is false.
3. Select it and click on the Wall command. This creates a single wall object (with four "walls", each on one edge of the base rectangle)
4. Select all of the four interior faces of the new wall (press and hold Ctrl while clicking on each face).
5. Then click on the Arch Space button on the toolbar
6. **Expected**: a space that fills the interior cube of empty space within the walls is created
7. **Actual**: the wall itself is converted into a space. Notice that the Boundaries property of this new space is empty.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40582 (Git) Snap 1327
Build date: 2025/03/06 21:29:39
Build type: Release
Branch: main
Hash: aa90da1541648cdb433d622e06b2d9ad27c9d509
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/95.1083/1
Installed mods: 
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.11
  * fasteners 0.5.37
  * fcVM
  * ezydxf
  * Arch
  * Assembly4 0.50.17
```

### Subproject(s) affected?

BIM

### Anything else?

Same behavior on 0.21 and 1.0

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-03-12:
Looking at the output of the Python console, it seems when selecting faces of a single wall this is called, which returns a single element (i.e. no boundaries):

```python
>>> obj = Arch.makeSpace(FreeCADGui.Selection.getSelection())
```

Whereas when selecting faces of multiple walls, this is called, which returns the subelements of each wall (i.e. the boundaries):

```python
>>> obj = Arch.makeSpace(FreeCADGui.Selection.getSelectionEx())
```

maxwxyz, 2025-03-12:
@paullee0 FYI regarding discussion for different algorithm in https://github.com/FreeCAD/FreeCAD/issues/6091

paullee0, 2025-03-13:
> [@paullee0](https://github.com/paullee0) FYI regarding discussion for different algorithm in [#6091](https://github.com/FreeCAD/FreeCAD/issues/6091)

Thanks for the heads-up, Roy_043's algorithm, right?  I have read that, yes it's good.

I pursued something TNP-Tolerant and no need to select the edges in a particular order though.  Need more spare time to refine and exposed them hopefully.

closed 2025-03-19 by yorikvanhavre