FreeCAD/FreeCAD #20144: [Core] Support for large files over 4GB

pieterhijma, 2025-03-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, FreeCAD does not support files over 4GB, see [this function](https://github.com/FreeCAD/FreeCAD/blob/53c4c2ef50870fd7a540b9511548b97a73f399b2/src/Base/FileInfo.cpp#L452).

### Full version info

```shell
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.40704 (Git)
Build date: 2025/03/12 04:23:27
Build type: Release
Branch: main
Hash: 53c4c2ef50870fd7a540b9511548b97a73f399b2
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
```

### Subproject(s) affected?

Core

### Anything else?

There may be other places that assume files of a maximum of 4GB.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pure-machinery, 2025-03-12:
Perhaps I'm not understanding it correctly but are you saying that the value gets converted from signed to unsigned and this way we limit to only max 4gb?

pieterhijma, 2025-03-12:
No, an `unsigned int` is 32 bit wide on many platforms, hence the maximum size is 2^32 which is 4GiB. So, this function reads a 64-bit value in the 32-bit variable. If the file happens to be larger than 4GiB, the reported size won't be correct. A signed int would be even worse because then the maximum would only be 2^32/2 which is 2GiB.

pure-machinery, 2025-03-12:
So basically we should have something like this:

```c
int64_t bytes = {};

// Rest of the code
```

maxwxyz, 2025-03-12:
Valid issue but I guess performance needs to be increased to be able to work with such files.

xtemp09, 2025-03-12:
True. Here is the proof.

<details>

```cpp

# cat stat_test.cpp
#include <sys/stat.h>
#include <iostream>

int main (void)
{
	constexpr wchar_t *filename = L"dadcrush.17.06.25.anastasia.rose.mp4";
	
	struct _stat buf;
	if (_wstat(filename, &buf) == 0)
		std::cout << "_wstat: " << buf.st_size / 1024 / 1024 << " MiB" << '\n';
	
	struct _stati64 buf2;
	if (_wstati64(filename, &buf2) == 0)
		std::cout << "_wstati64: " << buf2.st_size / 1024 / 1024 << " MiB";

    return 0;
}
# g++ -O2 stat_test.cpp -o stat_test.exe
# ./stat_test.exe
_wstat: 1114 MiB
_wstati64: 5210 MiB
```

</details>

pieterhijma, 2025-03-12:
> Valid issue but I guess performance needs to be increased to be able to work with such files.

Sure, but that's a different issue :) 
It just means it has not a high priority.

xtemp09, 2025-03-13:
Since _FreeCAD_ uses _Qt_, I would first use standard _Qt_ methods.

xtemp09, 2025-03-13:
> the complexity of constantly switching between libraries is no longer justifiable IMO.

+1. I would vote for getting rid of everything but _Qt_ wherever possible. In this case things behave predictably across different platforms with this issue included.