FreeCAD/FreeCAD #20145: Draft: Draft_Fillet not working at first  try

ErtugrulEge, 2025-03-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello,

Steps to reproduce:
1. Draw two connected lines (Draft_line)
2. Select two edges
3.Press Draft_Fillet
4.Enter a value and press enter it will give an error 

```
Edges are not connected or radius is too large.(Not true)
Fillet cannot be created
```

5.Click somewhere on the screen to deselect edges (or try with applying fillet tool without deselecting edges)
6.Select again two edge
7.Press Draft_Fillet
8.Enter the _same value_ and press enter, it will work 

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Turkish/Turkey (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/81.5973
```

### Subproject(s) affected?

Draft

### Anything else?

There is a related issue and tool has not an OK button
https://github.com/FreeCAD/FreeCAD/issues/14728

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-12:
Please add an example file.
@Roy-043 FYI

kaiwas, 2025-03-12:
I would also like to add that rounding is not confirmed by the Enter key on the numpad.

closed 2025-03-12 by Roy-043

kaiwas, 2025-03-12:
> I am not able to reproduce that with the information given here. Maybe [#20151](https://github.com/FreeCAD/FreeCAD/pull/20151) also fixes that problem? FYI: the task panel used here is also used for Draft Circle.

Probably the current algorithm of the tool does not imply work for vertices in multilines.
But choosing one vertex in a multiline (on the left) is sometimes more convenient than choosing two edges with the Ctrl key pressed

[image]

> Note that Draft Wires have a Fillet (and Chamfer) property.

Yes, I saw this parameter, but it works for all vertices in the multiline (right). And sometimes you need to make rounding only for some vertices (left).

[image]

In general, I accept that this tool has many limitations. It could be more convenient.