FreeCAD/FreeCAD #20157: Hovering over elements in imported geometry is very slow, so is orbiting around those elements.

drwho495, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

https://github.com/user-attachments/assets/39cff175-be5f-4b46-885e-5ad633e4e0e6

^^^ This video shows the current issue, whenever hovering over edges and faces in a imported step object, it runs really slow. It also lags when you try to orbit, meaning when you press right mouse button the program freezes for a few seconds before you can start orbiting.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/161.377/1
Installed mods: 
  * Render 2024.12.15 (Disabled)
  * ArchTextures
  * Assembly3 0.12.2
  * CadbaseLibrary 2.0.0
  * Behave-Dark-Colors 0.1.1
  * SearchBar 1.0.2
  * WebTools 1.0.0
  * Manipulator 1.5.7
  * POV-Ray-Rendering
  * cadquery_module 2.1.0
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * freecad.gears 1.3.0
  * ExplodedAssembly
  * fasteners 0.5.37
  * cadquery_module.backup1741030188.0603402 (Disabled)
  * FreeCAD-Telemetry 1.0.0beta
  * QuickMeasure 2022.10.28
  * FreeCAD-Ribbon 1.8.0
  * assembly2
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-03-14:
Can you share a file where this is as bad as in your video?

hyarion, 2025-03-14:
> > Can you share a file where this is as bad as in your video?
> 
> @hyarion In my experience this will happen with pretty much any imported STEP file.

"Pretty much" adds ambiguity to the issue. If you can supply one then we know for sure that it is an issue with that file for you.

It would also be interesting to know if this is an issue if you restart in safe mode (help->restart in safe mode)

srounce, 2025-03-14:
There is a step file [here](https://www.thingiverse.com/thing:4441537/files) (`Bondtech_BMG_+_E3D_V6_Volcano_Direct_Drive_Mount_REMIX.step` at the bottom of the list) which will be fine in terms of translation/rotation of the viewport so long as you don't hover over it, once I hover it slows to a crawl.

I'll try safe-mode in a bit and report back with results.

hyarion, 2025-03-14:
> There is a step file [here](https://www.thingiverse.com/thing:4441537/files) (`Bondtech_BMG_+_E3D_V6_Volcano_Direct_Drive_Mount_REMIX.step` at the bottom of the list) which will be fine in terms of [...]

I don't see any slowdowns with that model on my machine (macbook from 2021). Selection, hovering, rotation, everything works as expected. I don't have a way to count FPS, but it is at least 10 times faster.
```
OS: macOS Sonoma (14.5)
Architecture: arm64
Version: 1.1.0dev.40625 (Git) Conda
Build date: 2025/03/12 15:01:44
Build type: Release
Branch: main
Hash: 4cf3566cea3e77f7b34851a3e9469855bfec9080
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

closed 2025-03-14 by maxwxyz