FreeCAD/FreeCAD #20159: Fill light is not applied at create or open document.

oficinerobotica, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The "Fill light" is not applied in the 3d view when a new file is created or a existing file is opened. In the video bellow I intentionally set up the light color to green to illustrate the problem. Re applying the preferences enables the fill light in the 3d view

https://github.com/user-attachments/assets/4a79d1ad-4702-4ea8-b073-58b35ec4c0df

   

### Full version info

```shell
OS: KDE neon 6.3 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * fasteners 0.5.36
  * PieMenu 1.11.2
  * OpenTheme 2024.9.1
  * DisplayTools 1.0.0
  * freecad.gears 1.3.0
  * sheetmetal.backup1729482375.2960367 0.5.1 (Disabled)
  * sheetmetal 0.7.5
  * Curves 0.6.54
  * Assembly3 0.12.2
  * Gridfinity 0.8.0
  * Assembly4 0.50.15
  * CurvedShapes 1.0.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-03-13:
I think it's more than just what you've reported, if I apply any color to the Main and Back lights, restart FreeCAD and open or create a file none of the colours are applied unless I go into Preferences again:

[image]

 
[image]

Syres916, 2025-03-13:
Looking at `void View3DInventorViewer::init()` it does look wrong as the colours are hard coded and there's nothing obvious are to looking up the parameters set in Preferences with regard to the 3 lights.

oficinerobotica, 2025-03-14:
> I think it's more than just what you've reported, if I apply any color to the Main and Back lights, restart FreeCAD and open or create a file none of the colours are applied unless I go into Preferences again:
> 
Yes , you are right, and another strange behaviour is that the colors will be applied only once. That is , apply lights colors -> close preferences -> reopen preferences and change lights -> no change will be applied anymore until a restart

Syres916, 2025-03-14:
@oficinerobotica I used a new branch in the comment above to kadet (the _this_ link) which _hopefully_ fixes the Preferences issues as well. If you can compile then feel free to give it a test.

Syres916, 2025-03-14:
@luzpaz could you please tag this as a type:Bug, Status:Confirmed, Topic:User Interface and Status:Solution Proposed, TIA.

kadet1090, 2025-03-15:
> Looking at `void View3DInventorViewer::init()` it does look wrong as the colours are hard coded and there's nothing obvious as to what's looking up the parameters set in Preferences with regard to the 3 lights.

Hmm, that is strange - is it maybe due to https://github.com/FreeCAD/FreeCAD/pull/19995? Because it seems like observer is not correctly configured. @hyarion can you check?

hyarion, 2025-03-15:
> Hmm, that is strange - is it maybe due to https://github.com/FreeCAD/FreeCAD/pull/19995? Because it seems like observer is not correctly configured. @hyarion can you check?

Could you give me a hint of what I missed?

kadet1090, 2025-03-15:
> > Because it seems like observer is not correctly configured. @hyarion can you check?
> 
> Could you give me a hint of what I missed?

https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src%2FGui%2FView3DSettings.cpp#L98-L112 here you are triggering on change on group that you created, but you never attach observer to it on here: https://github.com/FreeCAD/FreeCAD/blob/9a6a98f824436656cae242ae26b043f78411ebba/src%2FGui%2FView3DSettings.cpp#L53-L64

Syres916, 2025-03-19:
@hyarion any news?

hyarion, 2025-03-20:
> @hyarion any news?

I tried to register the observer as mentioned by @kadet1090 but I can't get it to work.

I've been discussing it with him and he will look into it this weekend.

PR #20517 (2025-03-29): Gui: Fix saving light sources
As it seems the dialog saved settings in old place and because listener listened for the old place also the lights were updated only when applied through settings. Now it should be correct.

Fixes: #20159
closed 2025-03-29 by chennes