FreeCAD/FreeCAD #20170: BIM: Arch_Level does not add selected objects to it when it's created

furgo16, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD, switch to the BIM workbench
2. Create a wall
3. Select the wall
4. Click on the `Arch_Level` command button in the toolbar
5. **Expected**: the level is created and the wall is added to it
6. **Actual**: the level is created. The wall is NOT added to it

This looks like a regression, as it was working as expected in 0.21. It might be a small thing (an extra manual step to drag and drop objects into a level), but it's the sum of extra small things that hinder productivity.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40711 (Git) Snap 1337
Build date: 2025/03/12 16:27:41
Build type: Release
Branch: main
Hash: 2ad23cec23de917dab14f08fa6232410169a4ed2
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-13:
I can confirm the behavior. It is the same in FreeCAD 1.0.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073

```

kaiwas, 2025-03-13:
> This looks like a regression, as it was working as expected in 0.21. It might be a small thing (an extra manual step to drag and drop objects into a level), but it's the sum of extra small things that hinder productivity.

In version 0.21 it was not more convenient. There the object also appeared at the bottom of the tree.

But there is a way - an active group. That is, in the beam manager you need to make the level active.
In addition, the level has a property. However, for some reason it is located in the "View" tab.

Video demonstration.

https://github.com/user-attachments/assets/d9b4321e-ff11-4988-b66b-d1dea2d62393

In addition, Roy043 made an excellent "add-on" that allows you to conveniently insert objects by copy-paste.
Details at the link.
https://forum.freecad.org/memberlist.php?mode=viewprofile&u=22936

I really hope that this innovation will soon appear in the main version of the program.
Video demonstration. In addition to the correct appearance of objects, we can also conveniently insert objects by copy-paste. Video demonstration.

furgo16, 2025-03-13:
> In version 0.21 it was not more convenient. There the object also appeared at the bottom of the tree.

I think we might not be talking about the same thing. This is what I refer to in this issue. It used to work like on the video on 0.21 (walls included in Floor). In 1.0 onwards, selecting the walls and clicking on the Arch_Level button as per the issue description does no longer include them in the Floor/Level.

https://github.com/user-attachments/assets/ec6afa72-a555-43a9-a90b-9295e03e7e8c

kaiwas, 2025-03-13:
This is really interesting. Because other objects "wrap" the original. For example, you can make a column out of a wall.

closed 2025-03-17 by yorikvanhavre