FreeCAD/FreeCAD #20175: BIM_Classification window 'Cancel' button not working

semhustej, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- open BIM_Classification window
- hit `Cancel` button
- nothing happens

I am not able to reproduce in FreeCAD 1.0, seems like a regression.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical/physical DPI: 96/92.6073
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-03-13:
I can try to fix if approved.

Syres916, 2025-03-13:
Is this not a duplicate of #19857 , if so, Yorik has already raised a PR?

semhustej, 2025-03-14:
> Is this not a duplicate of [#19857](https://github.com/FreeCAD/FreeCAD/issues/19857) , if so, Yorik has already raised a PR?

Yes it is. I got confused with all the BIM windows :).

Closing the issue.

closed 2025-03-14 by semhustej