FreeCAD/FreeCAD #20182: Freecad gets stuck on this File Navigation window.

leoheck, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is the second time since yesterday that Freecad has gotten stuck on this File Navigation window. 
[image]

The last loglines are after having to kill the software:

[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40655 (Git) Conda AppImage
Build date: 2025/03/10 16:51:09
Build type: Release
Branch: main
Hash: f311bfc839a4c00eff9bfb294c4ec9f670410296
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.18-dev
  * sheetmetal 0.7.21
  * DynamicData 2.73.0
  * Curves 0.6.59
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.5
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-03-13:
Steps to reproduce it:

1. Open an existing file/project. 
1. Copy a Body (`Ctrl+C`), then `Ok`.
1. Open a new file.
1. Paste it.
1. It will ask to save the file first (why? Why? WHY? W.H.Y?)
1. Click `Yes`. (If I say `No` it does not paste anything)
1. Freecad gets stuck on the File Browser.

This is the popup on step #5.

[image]

leoheck, 2025-03-13:
Creating a new file, and saving it before of pasting the Body does not create this problem.

benj5378, 2025-03-13:
Could you please run in gdb, then when it freezes do <key>ctrl + c</key>, run command `bt full` and paste the output here?

maxwxyz, 2025-03-14:
does it also happen in safe mode?

leoheck, 2025-03-14:
I started to check this by trying to run it in Safe Mode since it was simpler.

I did this `About > Restart in Safe Mode`.

It closed itself and left this message on the command line but did not relaunch it.
My FreeCad was open 19h32min
```
/tmp/.mount_FreeCACHmKGf/AppRun: line 31: 48639 Bus error               (core dumped) "${MAIN}" "$@"
```

Tried it again, to make sure it would work for a fresh and recent run.
Now, it gave me this popup and I chose `Yes`.

[image]

Now, it restarted and then I got this

[image]

Still the same issue. 
The file browser opens the mouse turns into something spinning.. and the screen is stuck there in this situation.
I can move the popup window, but I cannot close it.
Actually, this time, I can close it on the window by clicking on the `X`.

After closing Frecad this time, it printed this
```
✦ ➜ Migrating Start Workbench to Start command... done.
```

Repeating this process, but not in `Safe Mode` I confirm I can close the File Browser by clicking on the `X` but the cursor spinning forever gives the wrong indication and also the `Save` and `Cancel` buttons don't work.

leoheck, 2025-03-14:
@benj5378 

Formated output

[image]

Textual

```bash
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/lheck/Applications/FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage...

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.ubuntu.com>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
(No debugging symbols found in /home/lheck/Applications/FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage)
"/home/lheck/Dropbox/deepx/Echo54/Ground-Sensor/Case/v1.3/Design/Parts/Battery_Holder.FCStd" is not a core dump: file format not recognized
(gdb) run
Starting program: /home/lheck/Applications/FreeCAD_weekly-builds-40655-conda-Linux-x86_64-py311.AppImage 
[Detaching after fork from child process 101371]
process 101368 is executing new program: /usr/bin/bash
warning: could not find '.gnu_debugaltlink' file for /lib/x86_64-linux-gnu/libtinfo.so.6
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 101377]
[Detaching after fork from child process 101379]
FreeCAD 1.1.0, Libs: 1.1.0devR40655 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

OSHAutoDoc workbench (v0.2.3) loaded
\

^C
Program received signal SIGINT, Interrupt.
Download failed: Invalid argument.  Continuing without source file ./posix/../sysdeps/unix/sysv/linux/wait4.c.
0x00007ffff7d107a7 in __GI___wait4 (pid=-1, stat_loc=0x7fffffffd220, options=0, usage=0x0) at ../sysdeps/unix/sysv/linux/wait4.c:30
warning: 30	../sysdeps/unix/sysv/linux/wait4.c: No such file or directory
(gdb) bt full
#0  0x00007ffff7d107a7 in __GI___wait4 (pid=-1, stat_loc=0x7fffffffd220, options=0, usage=0x0) at ../sysdeps/unix/sysv/linux/wait4.c:30
        sc_ret = -512
        sc_ret = <optimized out>
#1  0x00005555556716f3 in ?? ()
No symbol table info available.
#2  0x00005555555bef38 in wait_for ()
No symbol table info available.
#3  0x00005555555a1a62 in execute_command_internal ()
No symbol table info available.
#4  0x00005555555a1c6e in execute_command ()
No symbol table info available.
#5  0x000055555559410c in reader_loop ()
No symbol table info available.
#6  0x0000555555587efb in main ()
No symbol table info available.
(gdb) 

```

benj5378, 2025-03-14:
@leoheck we need to repeat the `bt full` in gdb with a debug build of FreeCAD

benj5378, 2025-03-14:
@oursland as mentioned

xtemp09, 2025-03-15:
@leoheck, could you share the file? I cannot reproduce.

Some files created in previous versions of _FreeCAD_ cannot be opened in the new one. The issue #19931 is a good example (possibly, #19949 is an another example).

leoheck, 2025-03-15:
@xtemp09 I cannot share the file. 

But my guess is that any file generated or edited before of current week will not be compatible with the weekly of the week. 

I am still working to have built that has the debug symbols enabled. I couldn't produce one yet on my system.

xtemp09, 2025-03-15:
From what I know about crashes, in most cases backtraces are useless without the reproduction steps, meaning that, a file must be provided to reproduce the crash.

> But my guess is that any file generated or edited before of current week will not be compatible with the weekly of the week. In this way those files are always incompatible.

Could you then provide a file to reproduce the crash?

masterxq, 2025-04-14:
I have probably the same problem.
How to reproduce:
1. Have one file with a link too a second.
2. copy the link too a third document that is new.

Video how to reproduce:
[freecad-#20182-2025-04-14_06.34.15.zip](https://github.com/user-attachments/files/19729631/freecad-.20182-2025-04-14_06.34.15.zip)

masterxq, 2025-04-14:
Also tried with:

```
OS: Debian GNU/Linux trixie/sid (LXQt/lxqt/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.5731/1
```
still happens.

masterxq, 2025-04-15:
@maxwxyz I believe the "Needs confirmation" tag and possibly the "Needs test on dev version" tag can be removed.
The issue isn’t particularly critical, since there’s a simple workaround: pressing the "x" in the dialog, saving the file, and trying again.
That said, users might not realize the "x" is still responsive, which could make this workaround hard to discover.
Still, it seems like a relatively easy fix and definitely an annoying bug.

FEA-eng, 2026-10-21:
> [@FEA-eng](https://github.com/FEA-eng) It has been my experience that this was fixed with v > 1.1.0. I had installed the dev version before the release and it was fixed then.

@leoheck Can you confirm that it's fixed for you too?

closed 2026-10-21 by maxwxyz