FreeCAD/FreeCAD #20183: Sketcher: Constraints need better helps

fester225, 2025-03-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am accustomed to working in PART. SKETCHER appears to be a different kind of drawing system requiring constraints, something automatically dealt with in PART.

I am currently attempting to create a screw thread through Sketcher. The profile and helix are pretty much completed. Of course I am left with attempting to satisfy the constraints. There are many tutorials describing what constraints are and how to apply them. **So far I have not found any descriptions of how to determine which constraints have not yet been satisfied**, even the Discord community seems to be having difficulty with the problem.

In the Solver Constraints box, the descriptions should be on by default. The descriptions should include the component names/numbers as well.

The Elements box should provide indications of which constraints have yet to be satisfied. The red/green/white lines/dots should probably mean something. This section should directly tell me what remains to be done.

https://forum.freecad.org/viewtopic.php?p=816605#p816605

### Full version info

```shell
OS: macOS 15.3.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * AnimationFreeCAD 1.0.0
  * pcb 6.2023.1
  * offline-documentation 1.0.0-alpha
  * kicadStepUpMod 11.4.4
  * fasteners 0.5.35
  * AirPlaneDesign 0.4.1
  * Help 1.0.3
  * Assembly4 0.50.15
```

### Subproject(s) affected?

None

### Anything else?

It appears other more experienced, but non-regular, Sketcher users are having the same problem. The DoFs box is capable of telling me how many constraints are left to be satisfied. Why can't it tell me which ones they are?

The responses from the forum which attempt to 'help' are mostly from gurus who think I shouldn't be using PART at all. Unless these gurus are willing to be available 24/7 to answer constraints questions online, having better, more descriptive helps would be a better solution.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-03-21:
Usually the parts of a sketch are rendered in light green which are already constrained. The remaining geometry can be constrained by a combination of various constraints, so I'm not sure how the application should suggest you to constrain it.
@FreeCAD/cad-working-group any idea, is this a valid feature?

pierreporte, 2025-03-28:
Sketch geometry can be constrained in a lot of different ways. I think it would be very hard to go beyond notifying the user which part is constrained or not. There are issues with vertex that are part of the geometry and not independent point to push this as far as it should be though (but the new sketcher style proposal does include this).